polyend/TrackerBetaTesting #440: Ability to assign Sample Root Key

meteorist, 2020-11-29:
It would be really nice if we could assign the correct root key for a sample so that samples in different root keys would correspond to the proper key on the keyboard and play the correct note.

A simple example:
If I record a sample played in the root key of E and then play a C on the keyboard it will actually be playing the sample in E . This becomes a rather annoying problem when editing a pattern and the notes played are not actually what they seem to be.

Inuitively I would be looking for this feature in Sample Playback.

modulationsource, 2020-11-29:
In the Instruments Settings there is the "tune" Parameter. 
So if you have a sample in E, just set tune to -4 (semitones) and it will play a C at the C Note.

meteorist, 2020-11-30:
I realize now, that I have probably seen that parameter before. Setting the root note in note letters rather than tuning in semitones would be slightly more intuitive though.

Conceptually I think that tuning would be more fitting to do in cents or something similar.

jakubhlavaty, 2022-04-07:
+1 for being able to set actual/root key for samples.
Tuning seems to be more of a workaround for this feature and a fiddly one at that imo.

Looks like this could be easy to implement too.
I assume the tracker already "knows" internally what key it uses for the samples.

So I would vote to
- add a "default key for samples" setting (not to be confused with pad root note)
- add an instrument parameter for the sample root note (e.g. "C#3") - i.e. "I know my sample is C#3, so please play the original/not-pitch-shifted sample when that particular note is played"