polyend/TrackerBetaTesting #441: LIve record in song mode

keith-macleod, 2020-11-29:
This would be a very powerful feature for composition. Translating a long melodic or thematic idea over a series of patterns is tedious and time-consuming process. If one could create the pattern sequence and then record while playing them back in order, this would be a very liberating workflow.

meteorist, 2020-11-30:
I think I read that something similar is in the beta firmware 1.3. The ability ro record while a song is playing. I'm not sure what the memory limitation is for Tracker but you would probably use up all the memory pretty quickly. After all Tracker is a sample player not a full fledged DAW.

keith-macleod, 2020-11-30:
@meteorist I'm referring to pattern recording, not sampling, yes you can now sample while playing a pattern, which is great.

Transvaluator, 2020-12-01:
@keith-macleod I wasn't aware one can sample while playing a pattern now, even on the latest beta (1.3.0b1). How does one go about doing that? When I go to sample record it stops the pattern.

keith-macleod, 2020-12-01:
@Transvaluator, just hit play while in the sample record screen. I think, however, that it's only midi events which play back, for reasons I don't claim to fully understand. Still very useful for resampling external hardware.

Transvaluator, 2020-12-01:
How interesting @ only midi events! Thank you for the info