polyend/TrackerBetaTesting #442: [1.3.0] n – Random note does not respect scale.

PaulAntioche, 2020-11-30:
I love your project and I find your approach and your openness really great.
well done.
in fx step effect
n – Random note 
does not respect ”the global musical scale selected for the pads in Config” as indicated in the instructions. I have already noticed this but now here is a demonstration video. We notice that the notes played are chromatic whereas the chosen game is ”yo” (C D F G A)
in this zip: a video.
bye and thank you

[tracker n.m4v.zip](https://github.com/polyend/TrackerIssues/files/5615352/tracker.n.m4v.zip)

MortalJS, 2022-02-20:
Hi @PaulAntioche 
Can you check if the problem occurs in firmware 1.5?

PaulAntioche, 2022-02-21:
No, now it's ok!
great!

MortalJS, 2022-02-22:
Closed. Fixed

closed 2022-02-22 by MortalJS