polyend/TrackerBetaTesting #445: Allow slower BPM (down to 20) for ambient musicians who prefer compact compositions.

Madrayken, 2020-11-30:
Hi folks,
I'm using the Tracker to write ambient music, and prefer super low BPMs. Any chance we could make the lowest BPM 20, rather than the current value?

Just-Peter, 2020-11-30:
I requested this also a few months ago. This was the answer of Jacek : 

''_At the moment, the lowest BPM in the Tracker is 40 BPM (because of the AMP envelope length).
But there's an option to overcome it easily for what you're trying to achieve - the step Tempo Fx will allow you to go down to about 10 BPM. So you should be all set :) We've released a new firmware for the tests today, which also slows down all the LFOs (besides AMP) to 128_.''

Nice advice, I think.

Madrayken, 2020-11-30:
Thanks for that! It looks perfect. 
(And apologies for the repeat 'enhancement' - I did search, but apparently failed to do it with any skill)