polyend/TrackerBetaTesting #449: Metronome in Sample Record

Transvaluator, 2020-12-01:
Hi Polyend!

Wanted to say first off, thank you so much for continuing to work on this awesome device. It is so power!

I was thinking about some ways to smooth out the workflow for musicians who like to live record/create segments and samples to integrate to sequences, ala the live record function on the SP-505. On that device, you can continue playing a pattern while recording new samples (or resampling) from line in. I often added basslines or synth licks on top of running samples to build my tracks iteratively. 

Currently, pattern playback has to stop while recording live samples. I imagine that if you could have prevented that you would have, as it seems non-trivial to change. With that in mind, a possible workaround that may be a lower code-effort would be adding a metronome to the sample recorder. This would allow us to have some kind of guide for when we create live samples so we know it will fit into our pattern properly.

tfspsound, 2020-12-01:
I hadn’t tried live recording over anything yet on my Tracker but I’m surprised to hear I can’t listen to what I’m working on while I record something. I would also find a recording metronome a useful feature to have so long as it stays off the recording. The option to have a metronome in play mode would be useful too

alexbuga, 2020-12-04:
I was about to open an issue regarding playing the pattern while recording a sample.
I know it’s probably a hardware limitation, but imagine this scenarios:

1) I want to play my beat and record some guitar lick. I need to hear the beat for timing and ... inspiration.
2) I’m sending midi to a synth and I’d like to sample that back into the tracker.

bichan17, 2020-12-05:
Having the pattern play during sample record would be so helpful for adding performances of other instruments into the tracker. At least adding the option to hear the metronome would be a huge help too.

jauhoss, 2021-02-11:
Totally agree with these enhacement ideas. Got my tracker couple of days ago and it was really confusing how the metronome works in sample record mode. 

1. Most important would be to make metronome audible also when monitoring is enabled. currently it's pretty much useless when using Tracker as standalone without external mixer etc what would make it possible to monitor with metronome.
2. Metronome on/off button could be in sample record page directly
3. would be really nice to have possibility to play pattern while recording, if technically possible.

soulfish, 2021-03-20:
This would be great indeed, just tried to record some guitars on my PT and was surprised I could not hear a metronome in sample record (thanks @bichan17 for the trick, I'll try that, as I want to record acoustic guitar I will still hear what I record). If there's a hardware limitation, maybe a visual metronome/flash could still help ?

Goldfasan, 2021-08-14:
Dear Polyend Team,
I also agree with this suggestion.
Another solution would be to allow monitoring of the playing pattern/song while recording samples, if the hardware allows such tasking.

demolen, 2021-12-01:
I want it too

ArneDal, 2022-02-09:
Same here.

Looove the tracker (1.5) but coming from the Octatrack, i find it strange that there should be limitations about the metronome? I have to do workrounds when i want to sample external synths.

So yeah,
- possibility to have metronome on when monitor is on would be awesome!
- possibility to hear the pattern when recording would be extra awesome!

Thanks

breskar, 2022-05-09:
I was looking around for real time recording options and found this discussion.
I'm really missing real time recording while monitoring a pattern playback too.

gzavo, 2022-07-22:
I stumbled upon this discussion. It is a bit sad, but if it is a hardware limitation, a visual metronome on the sample recorder page would still be useful (as @soulfish suggested), and I assume there is no technical problems with that.