polyend/TrackerBetaTesting #451: Font change :D

alexbuga, 2020-12-03:
Hey there,
I know the font used in the UI is the least of your problems :D but it really bugs me. I know it has a retro aesthetic but it would be nice if you could provide a couple more choices. Your brand font for instance looks really nice maybe you can use that one, or even better... a /Fonts folder :)

It would really look more polished with a more modern font.

Thanks for listening :)

hepto, 2021-02-20:
Adding to this, given the low res of the screen i'd actually prefer font smoothing / aliasing to be disabled as it's a bit too blurry for my taste.

sehmaschine, 2021-04-16:
+1 on this. it currently looks like a monospace font with failed kerning (esp. with the "r"). there are many good fonts out there which are made for screens in the first place. I'm sure that this is not your primary concern, but given the overall aesthetic of the polyend brand, it would be worth considering IMHO.

MetabaronCZ, 2021-09-25:
Its time for pixel art graphics. You can render in half resolution and save some resources for app logic :).