polyend/TrackerBetaTesting #454: Simple Inversions of Midi Chords Command 0

modulationsource, 2020-12-03:
Hi, 
now that you have implemented that great Midi Chord feature...
I think the choice of chord is fine, but I miss simple inversions for the "neighbour" triads at least.
This would be 038, 049, 058, 059
Addition of these 4 triads would allow a lot of additional musical possibilities.

Just-Peter, 2020-12-04:
That would be a great improvement, +1

free-ghz, 2022-04-28:
Amazes me that this wasn't implemented.

Epitrochoid, 2022-05-15:
Just want to add that this really adds a lot of power to the chord command without noticeably cluttering the available options. Having triad inversions opens up pretty much all 4 note voicings (including drop 2) along with with polychords using only two tracks.

Examples

- Cmaj6 drop2 -> A + C 049
- Open Cm7 -> C + A 038
- Dsus "So What" voicing -> D + G 059
- Ddim -> D + G# 059
- Cmaj7 -> C 049 + D 059

free-ghz, 2022-05-15:
Also a missed opportunity to do one-step "key slides" (like on a keyboard if youre to play a G, you start by momentarily striking the F# then sliding off to the G). On LSDj where you can have negative chord offsets, i'd do a lot of `-1 0 0` on a single step to achieve this. Easy enough to do with only positive also. But then i need my `011` chords!

These are good for like trills and so on as well. Thinking about tracker arpeggio as only for chords is a mistake!