polyend/TrackerBetaTesting #464: render selection.. as loop!

gherat, 2020-12-05:
In order to re sample yourself and then equal slice your render result, “render selection.. as loop” functionality is needed. At the moment, it renders the selection as audio.. plus a whole range extra! Which makes it impossible to accurately slice the rendered result up again in equal slices.

jtkthx, 2020-12-21:
Fully agree.  Ricky Tinez on YouTube has a video where he experiences this exact use-case, where he has to cut and drag the reverb tail back into the beginning of the loop.

There's definitely value in the way it is now, rendering out the full reverb tails, but I expect more often than not people are going to want to render out loops.

colonel-blimp, 2022-01-17:
> There's definitely value in the way it is now, rendering out the full reverb tails, but I expect more often than not people are going to want to render loops.

* Related: #433