polyend/TrackerBetaTesting #465: Timestretching and or pitch to bmp locking

gherat, 2020-12-05:
This machine really really really needs timestretching or at least have an option to logarithmically lock pitch to bmp: so that say a sampleloop is 145 bpm and you slow tempo to 120 that the sample plays slower. At the moment I have to recalculate and set tempo’s for all loops which makes it no fun at all to combine loops of different bmp’s. Again, timestretching would be even better.

doelman, 2020-12-05:
Timestretching is there in the newest beta :)

gherat, 2020-12-06:
Nobody wants to loose an fx slot just for timestretching. You should be able to just type in the bpm of a loop and say wether or not you wish to timestretch it in the sample editor...

gherat, 2020-12-13:
We need realtime timestretching, tempo locked timestretching with independent pitch. So just give the bpm of a loop and it will stay in sync if you change your song bpm. Lije the octatrack amd ableton live..