polyend/TrackerBetaTesting #467: Perform mode in song mode

ketumatek, 2020-12-05:
Firmware v 1.3 b2

When you are in song mode and you click on perform, loop pattern is ON.

It will be very nice if, we can be on song mode, go on performance mode with out looping pattern to perform during the entire song and not only in a pattern of the song.

There is a a bottom resum performance (first on the left) which we could use to stop all the performance we did just before (put all the preset on off)
For the moment I don t notice it change something we I use it.

tfspsound, 2020-12-06:
I’m surprised that more people don’t find this an issue. Don’t people perform songs? Switching back and forth is not an option because going back into song mode kills the performance effect. So for example if you set the tune performance effect to -3 (eg) then to progress to the next pattern you have to go into song (back up to 0 in terms of tune) then back to performance to reset tune to -3. It just doesn’t work very well. Also if you set performance playback to reverse then leave to go into song mode it plays havoc with the pattern order.

discohead, 2020-12-06:
Just voicing my support this. Using Perform FX while the Song mode pattern arrangement continues on makes a lot of sense to me. I was disappointed to find that it's not possible. I'd love to see that change!

anemlife, 2020-12-10:
+1 About this! We definitely need this option. 

Also maybe a shortcut to make every track pattern go +1 or -1 in Perform mode would be nice! 
Performing is possible the way it is now but I don't have enough fingers to press every 8 tracks buttons + the Jogwheel at the same time when I need a transition ahah

VincenzoUser, 2020-12-19:
Just to give my full support and adherence to this request.
This limitation prevents launching effects on the fly during the song, forcing you to stop on only one looped segment of the song: in this way live usability is compromised, that is, you let the song flow with the effects setted at home and you cannot act live in real time on them (if you need to keep the song structure)...
The great value of the Polyend Tracker in my opinion is to allow active (not passive) live performance without a computer...
So this enhancement would be very important indeed.
It would be also doable, I think, because I learned that old firmwares already had it: it was removed only because it created some (I really hope solvable) problems...
Thanks in advance!

ambv, 2021-01-19:
Duplicate of #413, let's discuss there.

closed 2021-01-19 by ambv