polyend/TrackerBetaTesting #469: [1.3.0 beta 2] Can’t overdub if there’s a note without a note off command

TwistedGenetics, 2020-12-05:
Hi 

When overdubbing  notes won’t record unless the existing notes have a note off value. I have enclosed a video to show this bug.

https://share.icloud.com/photos/08obyKyZEmIERyjaCWwcJl4fQ

Thanks

ambv, 2021-01-19:
We need to figure out what the design here should look like. The idea behind not recording notes into tracks with open notes is to not cut them off if new notes can be recorded on a different track instead. This works as intended but I agree we are missing support for a case where the user wants to overdub regardless of the current track state. If you have suggestions on how this could work, we're happy to hear them.

TwistedGenetics, 2021-01-19:
For me if I want to overdub it’s to add notes to and existing instrument I missed the first time around. So if these new notes cut existing notes then so be it ( they would have cut if recorded in correct order the first time ). With regards to overdubbing new instruments on an existing track I  think that if a note is already occupied when you hit a new note then no note should be added, only empty spaces can be occupied?

XANi, 2021-01-27:
My take on improving it for live record:

* Hold shift during recording to get to overwrite mode (any note under will always be overwritten by current incoming note) + option in config to switch whether overwrite mode is under shift or by default
* Press delete during to remove any note currently under it. Holding it during live record just constantly removes the notes under it, adding any currently played by user.
* Press shift + delete to remove note currently under it *and* insert OFF if there is any note playing

That way it would be possible, if say you wanted to have another take at a part, just hold delete and play to replace it with new notes, or hold shift if you just wanted to change/add new ones

> We need to figure out what the design here should look like. The idea behind not recording notes into tracks with open notes is to not cut them off if new notes can be recorded on a different track [...]

@ambv simplest fix for that would be

* splilling if more than one track is armed
* adding if only one track is armed

Then if I want to say work on one track I can just disarm all other ones. Option to delete one under cursor would also be nice

bambrose, 2021-01-27:
I think it should be destructive overdubs.. I.e if you don't want to overwrite stop playing.
Playing and having notes not hit just feels like something is broken

XANi, 2021-01-28:
@bambrose I had the same feeling and was actually insanely confused at first when it happened. Then I noticed spilling notes so I unarmed rest of the tracks. Then was surprised that I couldn't add *anything*. Only after finding this bug I have realized it won't add because there was no OFF after the note.

PITmast, 2021-05-09:
Hi XANi
Finally I got what’s going on with midi live recording. So it looks like missing OFFs cause strange behaviors after erasing entire midi track during playback. Thanks!

I absolutely agree it’s worth to implement something to make work with midi more convenient because this gear can be the best mobile midi sequencer ever with absolutely unique advantage samples player.

Your ideas are great, maybe except keeping any PT key pressed during recording (both hands on keyboard). I absolutely support you. Poly, please consider it.

Regards

MortalJS, 2022-02-20:
Hi @TwistedGenetics 
Can you check if the problem occurs in firmware 1.5?

PITmast, 2022-02-20:
Hello everyone 
It’s still there in 1.5

regards

closed 2023-04-21 by MiroPoly