polyend/TrackerBetaTesting #472: 1.3.0b2 Midi Micro tuning affects all notes after microtuned step

dustinpero, 2020-12-05:
After microtuning a midi note, the microtuning carries onto the next note. If I tune C5 down -99, the rest of the notes following will also be tuned down -99 without the microtuning on that FX step. To get around this, I have to put M 0 on the next note to get back to the correct note tuning.

Microtuning Midi instruments should work the same as microtuning samples, by snapping back to default tuning after the microtuned step.

shuler323, 2020-12-10:
You're right... but.. microtuning for midi output creates specific situation, because it sends 'pitch bend' command, which is global for all notes in your slave device. Maybe it's safier to set it back to 0 by hand?

dustinpero, 2020-12-10:
Alright. I wasn't sure if this was working as intended or not. Thanks.

shuler323, 2020-12-10:
thanks for being alert :)

closed 2020-12-10 by shuler323