polyend/TrackerBetaTesting #473: 1.3.0 b2 Master Space parameter resolution

stevesomeone, 2020-12-06:
Possibly not considered a bug, but can we have better resolution on the Space parameter?
If you do an A/B comparison with the Bass Boost next to it, you will see what I mean.

* How often does this bug happen? (All the time since it was introduced)

* Does this bug happen in previous firmware versions? See above

* Does this bug happen in every project? Yes

shuler323, 2020-12-10:
i'm sure that it would be better to have better resolution, but range 0-8 is limited by piece of hardware that generates this effect and it can't be greater than this.

closed 2020-12-10 by shuler323

tilllt, 2020-12-28:
I am not sure, but I maybe there is a misunderstanding? IMO the parameters should be more _fine grained_ and not _more_. Setting space to "1" or "2" is not subtle at all, but extremely audible.