polyend/TrackerBetaTesting #474: [1.2.1] Pattern Play Mode FX crash

tfspsound, 2020-12-06:
Steps to reproduce

Enter Perform from Song with it playing. 
Set pattern play mode FX to Back
Go back into Song and hit Continue Song. 
The FX is still being applied so when it gets to the end (or start since it’s playing backwards) of the loop that is playing the play head freezes and if you leave it, it starts to jump between patterns whilst playing out the samples. 

Attachments and links

No

Other important information

* How often does this happen?
Every time. 

* Does this bug happen in previous firmware versions?

1.2.1

* Does this bug happen in every project?

Yes

shuler323, 2020-12-10:
@tfspsound  You're right... but.. microtuning for midi output creates specific situation, because it sends 'pitch bend' command, which is global for all notes in your slave device. Maybe it's safier to set it back to 0 by hand?

tfspsound, 2020-12-10:
Hi

I’m sorry I don’t know the in and outs of midi like you guys. I think the problem could be resolved by using a Pattern Play Mode fx to Loop within Performance FX. Then if Performance Mode worked by playing through the Song instead of looping as it does currently, with access to Loop we could decide when to loop without leaving the performance screen which breaks the continuity of the track. Also the tracker wouldn’t need to work out where it needs to be when loop is removed. But then we might be back where we started. Just thought I’d report it because it looked like a bug. I stopped it because it looked like it was getting quicker but someone else might leave it.(?) Thanks for looking into it. 

Sam