polyend/TrackerBetaTesting #478: Métronome new possibility

ketumatek, 2020-12-07:
Hi

For the moment I can use the metronome only in live recording mode.

It will be nice if , as it was ask in another tread, we have the possibility to choose were to use the metronome 

- Live recording mode
- Pattern mode
- Pattern mode recording 
- Live pattern mode recording 

In pattern mode recording it can be useful to known if you place your sample in a good musical way...espacialy for drums when you use slice.

So if you have the metronome during you put different slice of a drums you are sure that you place stuff in the right place against the time.
As this we have a reference.

math0ne, 2020-12-10:
This would be very handy for when you are laying down a loop, going from practicing something before you program it without hearing a metronome to hearing one when you hit record is quite jarring.

jaap3, 2020-12-27:
Probably related to #256

ketumatek, 2020-12-27:
Yes you are right seems to be a little bit the same...

demolen, 2021-12-01:
I second that. Metronome is an essential musical tool. It doesn't make much sense just to limit it to the record mode. Even in reality, one would not choose a metronome only when wants to record something.

closed 2021-12-02 by Jacknametrouble