polyend/TrackerBetaTesting #486: Roll control

tfspsound, 2020-12-09:
I was thinking it would be nice if there was a way to make your own custom rolls using a table with 2 adjustable values. 1: Number of repeats, and 2: Number of steps to repeat over. 
I don’t know if this is possible but it could help when you need the same sample triggered 2 or 3 times in the same step or to create odd numbers of beats over many steps.

gherat, 2020-12-13:
yes: speed, length and velocity ramp type. That’s all you need to make all the roll types instead of this “preset system”

thumperjack, 2021-11-27:
yes, if you had control over 1. number of repeats and 2. repeat resolution, and 3. volume ramp... roll could still be subdivisions with a high resolution and length of 1 step...

or if you stretched out the parameters to more than 1 step and lower resolution it can be really easy triplets, or quintuplets, triggered from one note. 

then you can micro time that one note and the whole group moves together. it gets really fun when resolution isn't locked to grid, octatrack has it as a number of repeats and re-trigger time ratio, you can find an exact grid resolution, or you can have it fire away in free time not locked to grid, or you can bump the resolution while it's ratcheting with an LFO, that gets really fun