polyend/TrackerBetaTesting #494: Suggestion: a faster way to change keyboard's octave

anemlife, 2020-12-10:
You always need to go to the config menu to change the displayed octaves for the keyboard, which can sometimes kill the workflow. 

What about using the Notes screen (the one that shows every corresponding notes in relation to pads when you keep the button) + Shift + Jogwheel to change the displayed notes/octaves?

That would really make it faster!

alexbuga, 2020-12-11:
Yeah. I would also love some quick switch options for Root, Scale, Layout while holding down NOTE.
It’s useful because you can quickly switch between whatever scale you wish and Chromatic (which is kinda mandatory if you wanna play slices)

pete223, 2021-06-12:
Would definitely make the Notes screen more useful if the options for Root Note, Scale and Layout were present on this screen.

I would suggest that holding down the Note key would bring up the Note Page and keep it open, then the following options would be available in the bottom tray and any change would update the notes instantly, which would also help in learning the different scales.

[image]

anemlife, 2021-06-13:
> Would definitely make the Notes screen more useful if the options for Root Note, Scale and Layout were present on this screen.
>
> I would suggest that holding down the Note key would bring up the Note [...]

Exactly what I was thinking about. It would make the note screen way more useful than it is now :)))

pete223, 2021-06-14:
> Exactly what I was thinking about. It would make the note screen way more useful than it is now :)))

Yup, at the moment the Notes Page serves a very limited purpose and I don't really use it at all.

jakubhlavaty, 2022-04-07:
+1 for fast octave switching. It is indeed a bit of a flow killer.
MIDI controllers usually even have dedicated buttons for that.
I'm kind of surprised that it's not there from the start.