polyend/TrackerBetaTesting #496: Suggestion: Tuner

anemlife, 2020-12-10:
I know this is a massive request but that would definitely be a crazy thing to add. When using the PT outside with headphones I don't have any solution yet for tuning my samples correctly and that is one of the things I'm really missing. 

(If any one got a trick for this let me know!)

alexbuga, 2020-12-11:
Just have a C3 piano sample or better... a sine wave at hand.
Load that one and put a step in the seq and then live play the new sample while in Instrument Params and set the tune. Works like a charm. It also trains your ears.

jtkthx, 2020-12-21:
A tuner would be useful for paring with analog gear, eurorack, etc.

witchbutter, 2021-01-03:
I would suggest further that a tuner in the record view would be helpful but more than that I would appreciate a tuner and pitch correction effect listed in the sample editor.  Right now I am copying samples and recordings into Live to pitch adjust them then copying them back to the SD card.

anemlife, 2021-01-04:
@witchbutter I second that! I'm definitely missing a tuner fx in Sample Editor to make the original sample match the keyboard + tuner without having to resample anything

also quick tip if you want to avoid using a computer to do this while there is no tuner option on the PT(maybe you already knew about this method but here we go): 

1) load a sample 
2) write a new pattern with only this sample playing in the tuning you need
3) select the note + length of sample wanted (shift + arrows)
4) go to the "more" menu --> render selection --> render and load (or render and then replace the first sample with the tuned one, as you will probably delete the first sample anyway)

I always leave a blank page for ideas + resampling in the page number 48 as my songs rarely go that long, so I can do these  resampling manipulations faster

mxactvtd, 2021-01-28:
If you do make a tuner, please also have a hz view mode like in Live or Reaper

theorize999, 2021-03-08:
Please implement this, I currently have over a hundred gigs of samples on my PT and not everyone who makes samples puts their pitch in the name, and it really becomes a chore always manually tuning samples. This would be one of the best workflow improvements that could be done right now. Not everyone has perfect pitch and shouldn't really be expected to, PT is billed as an audio workstation... it really should have a tuner in it.

rjvaleo, 2021-03-10:
another solution could be to put a simple sine wave generator in the
polyend so we could save out sine waves at the desired pitches and use them
for more than just tunings...
--

TWITTER: http://twitter.com/rjvaleo <http://twitter.com/#/rjvaleo>
SOUNDCLOUD : http://www.soundcloud.com/rjvaleo
TRANSELECTRONIC: http://www.transelectronic.net

0kubi, 2021-12-30:
Would really really love this, sometimes you just don't have tuning for a sample (or they're not a tuned instrument/microtonal) and being able to check what the tuning is would be great

Aisjam, 2022-03-27:
Doing a lot of single cycle instruments, I would like this too please.