polyend/TrackerBetaTesting #506: [1.3.0b3] Slices don’t play anymore in “Beat Slice” mode

alexbuga, 2020-12-12:
## Hi, sorry you had trouble!

### Steps to reproduce

I have a guitar loop chopped. The pattern plays only one slice. If I go to Sample Playback and tweak the slices when I play the pads, it still plays just one slice which you can’t change.

* How often does this bug happen? (every time, often, "it only happened once”)
Every time

* Does this bug happen in previous firmware versions?
No

* Does this bug happen in every project?
Will test.

LetsSeeIfWeSolveThis, 2020-12-12:
Jhmm, I have the samples, but now on the second octave, not the third. (C2 ; C#2 ...) 
Its the same problem?

alexbuga, 2020-12-12:
I can’t test it anymore. Rolled back to prev beta

alexbuga, 2020-12-12:
I tried what you say but still can’t play them.
Reverted to beta 2 for the moment.

ketumatek, 2020-12-12:
I try to make a slice on a beat on the last beta b2 firmware it was working well.
means slice begun on C3 and all the other slice was after.
Now with the new beta b3. It just play the last slice on all the pad...same as if I took an old slice I did before...
Same problem

prds100, 2020-12-12:
I have this problem too, fixed by adding slice fx on the tracks but not ideal, hopefully this is patched out quick.  Tried altering the octaves of the notes in the step but always fell on the last slice of the sequence (also in sample playback menu)

atav1k, 2020-12-13:
having the same problem with beat slice. only higher notes past C3, C2-B3 do not play.

sakke-makke, 2020-12-17:
Same here. I use equal slice command and it chops the sample nicely, but it plays only the last slice. Certainly needs a fix!

closed 2020-12-22 by alexbuga