polyend/TrackerBetaTesting #507: Feature Request: Use Copy/Paste in Sample Loader:

zepi8888, 2020-12-12:
## Hi, we're excited to hear your idea!

Before you start, did you check whether your feature was already requested?
Searching might save you quite a bit of time here.

### A new feature

Please tell us about your feature in as much detail as you can provide.
Vague ideas are less likely to be successful.  Concrete requests are
easier to estimate.

dustinpero, 2020-12-12:
I believe you can copy and paste in the sample loader. Sometimes you have to hit copy on the file for it to take, but I copy files and paste all the time in the sample loader window.

zepi8888, 2020-12-13:
The idea is to copy/past Instruments in the Sample Loader (in the right box) without having to find the original sample. Eg. the pointer to Instrument Nr. 6 > press Copy > move the pointer to Nr. 7 > press Paste. 
[image]

zepi8888, 2020-12-13:
Ok, sorry, I just tried the "press copy multiple times" trick and it worked. That's exactly what I was looking for. 
For some reason it seemed like it doesn't work first but it does now. 
Thank you guys and sorry for the confusion.

closed 2020-12-13 by zepi8888