polyend/TrackerBetaTesting #511: [1.3] Big latency issue

gherat, 2020-12-12:
There seems to be a really big latency problem when using midi (with use of a midi cable). It runs completely out of sync and is therefor unuseable in combination with other gear. The latency is around 20ms (!)

clickysteve, 2020-12-13:
Is this for MIDI output (clock + notes), or input?

gherat, 2020-12-13:
I encounterd the issue when running the tracker as a slave to ableton. When I do a -20ms clock sync delay in the preferences it runs ok.

sebpmtl, 2020-12-14:
I feel there's a latency issue in the machine itself, like there a slight delay when playing notes with the pads and not in midi either.

Adam-Vandal, 2020-12-18:
looks like this is reported long time ago. I'm also impatiently waiting for the fix :)
https://github.com/polyend/TrackerIssues/issues/92

shuler323, 2020-12-22:
try 1.3.0b4, it has implemented first sync delay compensation in PT

closed 2020-12-22 by shuler323

gherat, 2020-12-22:
Great! I’ll give it a go!

Adam-Vandal, 2020-12-22:
Looks like MIDI compensation now works only for incoming signal, not when Tracker is as master . :( Right?

shuler323, 2020-12-23:
@Adam-Vandal don't be sad! it will be implemented soon :)