polyend/TrackerBetaTesting #512: [1.3.0b] Export Song Stems - Various Issues including tempo inconsistency

clickysteve, 2020-12-13:
Following on from the issue relating to the [tempo of exported stems being inaccurate](https://github.com/polyend/TrackerIssues/issues/281), which was reportedly fixed, I have tried exporting song stems but run into the same issues, along with a few others:

**Issues:**

1. Rendering song stems is extremely slow, even for relatively short songs. For example, exporting https://www.youtube.com/watch?v=1jEE1_4CKL0, which comes in at a total length of around 3 minutes and 20 seconds, took about 2 hours to complete. The track was not especially over-bearing, over-loaded, or memory intensive from what I can gather. 
2. The process crashed on the first attempt without completion or warning. The first time I tried to export song stems, the process crashed at about 40% complete. There was no warning or indication that it had failed - the 'Exporting Song Stems' popup simply no longer appeared, and the export files were corrupted.
3. The song exports are named _SONG_S000*, and don't relate to the name of the project file, as they did in previous versions.
4. Most importantly, the tempo export is still not correct, and drifts over time. You can see this evident here, where the start of the stems begin before the bar marker near the beginning of the project:

[image]

yet by further on in the project, they are behind:
 
[image]

This is using the same samples, in the same positions, with no tempo adjustment in FX. 

It is further evident by the total length of the export stems, which all finish at different times (and not on a bar marker):

[image]

Listening to the stems together makes it very immediately clear that the tempo isn't accurate, as they aren't in time with each other - and are very badly out of time with the Master. 

Thanks

clickysteve, 2020-12-13:
I just tried the pattern stems export and experience the same issue.

1. Patterns don't have 'friendly' names relative to the project name
2. The tempo and length of the exported patterns is off, in some cases significantly. Example of two different patterns:

[image]

[image]

gherat, 2020-12-13:
There are major sync and timing issues with this machine to a point where it's unuseable with other gear. However you mustn't confuse the "tail" of an export with the timing issue. I think the export is taking into account that there is for instance a reverb tail or a longer sounding sample which it also renders so you won't loose sounds that play after the end of the bar..

However, I synced the device as a slave with midi and there was major latency, it ran next to other hardware which ran fine and in sync. It had a major delay and also "wobbling" timing.. I think the cpu of this machine can't handle playing steady and doing all calculations at once or something. Curious if they are able to fix this, 'cause it seems to me this is a huge problem..

thebradclay, 2020-12-13:
Same problems plus a few others that I just made a bug report for. Stem rendering takes forever and then they are so badly out of time that I might as well record each individually.

clickysteve, 2020-12-13:
> There are major sync and timing issues with this machine to a point where it's unuseable with other gear. However you mustn't confuse the "tail" of an export with the timing issue. I think the [...]

I hear what you are saying, but it doesn't apply here. Firstly, reverb and delay are exported as separate stems. Secondly, Secondly, exporting individual pattern stems at different lengths is no use. Thirdly, the stems are so badly out of time with each other (both in song mode and pattern mode) that it is audible. The 'master' export which is a mix comes out a different time as the stems - and if you play them back together they are out of time with one another. There is definitely a timing issue.

clickysteve, 2020-12-13:
> Same problems plus a few others that I just made a bug report for. Stem rendering takes forever and then they are so badly out of time that I might as well record each individually.

Right. I am thinking I'll just have to solo each track and record the channels individually - clocked to the DAW - like I have to do with the Game Boy. With 8 tracks that's really no use though. I am fairly disappointed, as the stem export is what made this machine so much more useful than the Digitakt. Hopefully this will be fixed soon.

doelman, 2020-12-13:
@clickysteve I tried that method (with clock coming from daw), recording each channel seperately. Even then there were massive tempo inconsistencies

gherat, 2020-12-13:
I get decent results when setting a midi clock delay of -20ms in Abletons midi preferences (for the midi output that runs the tracker). When I record patterns into Ableton I record it twice because the first hit is still late, but the rest of all hits are on the grid. The second time around the pattern all hits are in place.

clickysteve, 2020-12-14:
> @clickysteve I tried that method (with clock coming from daw), recording each channel seperately. Even then there were massive tempo inconsistencies

That's disappointing to hear. That said, DAW clock is pretty awful in general. I've got an ERM Multi-clock that I'll try slaving it to and see if that helps. Ideally it shouldn't have to though. Exporting stems should work.

shuler323, 2020-12-16:
Thanks @clickysteve for this report, we need your project to find out what causes those problems. 
As I wrote in some issue, export was implemented from scratch in 1.3.0b3 to eliminate some major issues, and of course it may have still some which was not found in developing process.

doelman, 2020-12-17:
I tried it as well on 1.3.0b3 and things are much better in sync (still not entirely sure but it's not totally messed up like it was before) but it took over 3 hours to render the stems for a 5.5 minute song ..

shuler323, 2020-12-17:
@doelman can you share your project?

doelman, 2020-12-17:
> @doelman can you share your project?

For sure, see attachments. Listened again together with the master file and I think it's in sync but yeah, it still took over three hours to export. 
[wise shoe.zip](https://github.com/polyend/TrackerIssues/files/5709251/wise.shoe.zip)

jaap3, 2020-12-17:
I'll try upgrading and exporting a track with the new firmware soon. I also had major timing issues and haven't really used the Tracker for a while because of this. Really hope it is better now, as the Tracker really is a dream come true, except for this export/sync issue.

doelman, 2020-12-19:
Did an export of the same project (just slightly changed) again, this time it took about 5 hours to generate the stems ... But they are in sync so I was kind of excited but now I found out it's mono and all my panning and lfo panning etc. is ignore..

clickysteve, 2020-12-21:
@shuler323 

> Thanks @clickysteve for this report, we need your project to find out what causes those problems.
> As I wrote in some issue, export was implemented from scratch in 1.3.0b3 to eliminate some major issues, and of course it may have still some which was not found in developing process.

Sure. This is the project that I referenced earlier. Export took around 2 hours for the stems, and they are all out of synch (both full song and patterns), even with the Master:

[old-fashioned amusement.zip](https://github.com/polyend/TrackerIssues/files/5724090/old-fashioned.amusement.zip)

Just to echo what @jaap3 said: The Tracker is a great device, and I really enjoy using it, but I've put any work with it on hold indefinitely, as without the ability to accurately tempo synch/export, then it unfortunately isn't a whole lot of use. I don't want to sink a bunch of time into projects only to discover that I can't do much with them, or that they are useless with new firmware upgrades (had this nightmare with Digitakt and Overbridge). Hopefully you manage to get this sorted out!

jaap3, 2020-12-21:
I upgraded to the current beta firmware and was not able to export my project at all (I only tried one project). I let it go overnight and in the morning it was still on the exporting screen. I unplugged the Tracker, rebooted and tried again. I let it go for the entire day with the same result.

The project I'm trying this with was started on the original firwmare (I guess 1.0, I can't remember). Maybe that has something to do with this? It plays fine however, so I'm not sure.

shuler323, 2020-12-22:
@jaap3 
> current beta firmware

please note exact firmware code

jaap3, 2020-12-23:
I just updated to 1.3.0b5 and was able to export. The stems start out fine but suddenly around 57 seconds in they get horribly out of sync. Strangely enough when I listen back to the master mix (master.wav) everything seems fine.

I've attached my project, maybe it can help you reproduce this:

[spoilt cabbage.zip](https://github.com/polyend/TrackerIssues/files/5737053/spoilt.cabbage.zip)

jaap3, 2020-12-24:
With 1.3.0b6 I am able to export a different project successfully, the stems stay in sync. The only minor issue is a tiny ~12ms delay at the start of each track (it's the same on each). This can be trimmed easily so the export is entirely workable.

Now I'm wondering what's up with the other project. Is is somehow corrupted in a way that only is noticeable when exporting stems?

I'll experiment with exporting some other projects.

doelman, 2020-12-24:
@shuler323 before I test it, do you happen to know if the issue that the stems are in mono is fixed in 1.3 or should I create a separate issue for that (if it still occurs)?

jaap3, 2020-12-24:
@doelman the files are stereo for me. I don't have any pans so not sure if stereo information is kept, but at least the files seem to have two channels.

Sadly I did have issues exporting another project. One of the tracks (track 4) goes out of sync about 30 seconds into the song. I've attached the project, maybe it helps:

[erratic dog.zip](https://github.com/polyend/TrackerIssues/files/5739878/erratic.dog.zip)

doelman, 2020-12-24:
@jaap3 yes they have two channels but the stereo information was gone. Will try again today but I hope it won't take five hours this time..

headtopmusic, 2021-01-01:
I am trying to export 2 projects as song stems that were made in firmware v1.2.2 through the 1.3.0 firmware. 

Every time I export using 1.3.0 the stems are inconsistent in length.  I left a project exporting for 12 hours and the result was a set of stems that don't match in length.  I tried reverting to 1.2.2 but the exported files drift out of time gradually similar to the issue @clickysteve reported.

As an alternative, my next step is to solo each channel and record manually into my DAW. This method is flawed though. The Send channels will likely not match the recordings of my Track 1-8 stems as there are moments of randomization FX that change on each playback.  I'll either record each channel with the Send for Reverb and Delay audible on each recorded stem or accept that I can't use the sends in my track due to the broken export song stems functionality :(

I am using Ableton to mix and these export issues are causing me a headache. This rendering issue is becoming a pain and stopping me from making music. I like the Tracker but the rendering is a major flaw that needs prioritizing and resolving. I've spent the last 2 days scratching my head exporting and testing. Each round takes half a day.

doelman, 2021-01-01:
Yeah I agree @headtopmusic, this should be very high on the list at Polyend. I love the device but besides the crackles and ticks (that's another issue) this export impossibilities are quite a deal breaker.

thebradclay, 2021-01-01:
I agree. I love the machine but I have literally spent hours trying to figure out how to get the stems to work. Randomly getting lucky sometimes here and there but it's not worth it for me to render.  

I also wonder what is actually going on with the rendering. I sat and render a couple of short stems I made just to see. First, the Tracker audibly plays each stem while rendering. Which makes me scratch my head. Why? Also, why would it take so long. For one song I recorded all the stems manually, including the sends and it took me about 45 mins. I rendered them and it was somewhere around 3.5 hours. Why?

The other thing I find so frustrating with having to solo and record is when I have a tempo change. If you are using the T(empo) FX on a step it affects the pattern. However if you are recording a stem, you have to put that T(empo) affect in for each individual channel. Problem is if you happen to have both FX used where you need the T(empo) FX to be. Which leads me to wondering if some of the problems people are having with stems being so messed up comes from the T(empo) FX. Like I said above, during a render the Tracker plays each stem out while doing whatever it is doing, and if it isn't reading all fx lanes while doing so then the stems will be out of sync. Also if that is the case then rendering is going to be an issue considering the T(empo) FX  affects the entire pattern but when rendering individual stems it's a problem. 

Hopefully that made sense.

doelman, 2021-01-01:
@thebradclay I've never used the tempo FX so far but still had problems with the stems. In the newest firmware it doesn't audibly play the stems anymore during export, but it now takes forever to finish the export (>4 hours for a 5 minutes song)

headtopmusic, 2021-01-01:
@thebradclay Interesting theory around FX on each track causing things to shift out of sync, That said, I haven't used the Tempo FX at all in any of my tracks much like @doelman so it goes beyond Tempo FX. 

I tried cutting the length of the track as I wondered whether the Tracker was struggling to render long tracks. In both the full length 6 min track and the half-length 3 min track, the .wav for track 6 is 2/3 the length of all other stems. Regardless of track length, the length of stems remains relative to one another. A set of stems with a consistent length and tempo is what's needed.

This has me really puzzled. In Ableton, some of the .wavs have warp markers peppered throughout with random tempos assigned to each.

These results are all happening with 1.3.0.

thebradclay, 2021-01-01:
@doelman Huh, I could have sworn it was still playing them...but you know what, they did release two betas and then the firmware within a few days so I might have been on a beta when doing that. 

Either way, the problems still exist and agree with you all. I'm not bothering exporting until I know for sure it's fixed. Hopefully soon. 

@headtopmusic Agreed it for sure goes beyond the T FX. I've tried several little different pattern renders/stems etc and things are just way messed up. I use Ableton as well, and yeah I noticed all kinds of fucky things in the wavs.  

Also, someone above mentioned drift while recording individual stems into their DAW. Same issue here as well. This is such a huge waste of time trying to just get the stems out, and I have to have them because I am using the Tracker to control my modular and other stuff that runs to my mixer. Trying to get them to run into the 

I just want to help get this fixed because out of all the gear I've bought over the years, this is the one that I instantly gelled with.

ancipital, 2021-01-04:
Just to confirm, this is is still hosed in 1.3.0 release.  Rendering is incredibly, mind-blowingly slow too- a fraction of realtime.

I imported the stems into Live with no warping and the drums were out of time after a few phrases. I tried un-muting the master stem output to compare, and you could hear it phasing with the others from the very start. Something remains broken here- which is a real shame, if the output isn't usable.

clickysteve, 2021-01-05:
> Also, someone above mentioned drift while recording individual stems into their DAW. Same issue here as well. This is such a huge waste of time trying to just get the stems out, and I have to have them because I am using the Tracker to control my modular and other stuff that runs to my mixer. 

Yep. Trying to troubleshoot is very difficult when each export takes hours and hours. I could live with the ludicrously long export time if it was all perfectly in time, but both issues combined are a killer.

thebradclay, 2021-01-05:
I'm going to try something tonight just to so there is more data. I'll make a short 2 or 3 pattern song with some tempo changes and see how long it takes to export and then see what the sync issues are and upload them.

ancipital, 2021-01-05:
OK, on firmware 1.3.0 - release version. I am having a hard time making a project that doesn't suffer from obvious tempo drift in the exported stems. It's hard to test as thoroughly as I'd like as it's incredibly slow, many times slower than realtime.

I have included a small project to reproduce- it was made from scratch under 1.30 release. It doesn't have any weird tempo changes, the whole thing remains at 170. The export drifts markedly after a very short time.

thebradclay, 2021-01-05:
Okay so here is a small project that I just put together that consists of 2 patterns using 3 channesl, 3 samples with tempo changes. I've included the project folder, rendered stems for the 3 channels and the song .wav file. None of the tempo changes were output to the stems or the song. 

[PT Crash Tests.zip](https://github.com/polyend/TrackerIssues/files/5772016/PT.Crash.Tests.zip)

leverkissing, 2021-01-05:
I'd like to contribute that I went for the tracker because it was supposed to offer Stem export and I am quite disapponited that this still does not work and also seemingly has no priority for Polyend - instead we now can play Super Mario on a sequencer...wtf.

clickysteve, 2021-01-05:
I hope Polyend will prioritise this issue, as it doesn't appear to be getting the same attention that other, new bugs or features are. That is pretty disappointing, as it's a core function of the device.

doelman, 2021-01-06:
@shuler323 Just to make sure it won't be overseen, the following issues are also related to exporting and especially #580 is a big one for me as this makes the stems (if they generate at all) totally unusable whenever you do something stereo related since the stems are mono .. 

other one that has been a bug since 1.2 already: #586

thebradclay, 2021-01-06:
@doelman they just closed my report, #644 with zero explanation and the problem persists.

doelman, 2021-01-06:
> @doelman they just closed my report, #644 with zero explanation and the problem persists.

im sure that's a misunderstanding, they probably have enough information in this topic already as it basically describes the same problem.

thebradclay, 2021-01-06:
Said it was a duplicate.

clickysteve, 2021-01-06:
Don't worry. Duplicate reports are often closed so they can be better managed in a single place. :)

ancipital, 2021-01-07:
I'm sure it's not being ignored- it's an assigned bug and all. 

In any case, if there's no proper way to get an intact copy of your work out of the box, it renders all the other bugs and enhancement requests moot, so I am sure they're looking hard at this one :)

headtopmusic, 2021-01-07:
I have contributed to this issue with my problems exporting stems. I am still suffering from the stems being inconsistent in length with strange starting points. In Ableton, the stems all mismatch in length regardless of whether warping is on or off.

I've been hesitant to post this as I am still to finish recording all of the song stems out individually since my breakthrough but thought I would share as getting the time to record them has been hard :) 

After recording the stems manually and finding out the hard way that they all drift out of sync and suffer from mistiming I spent hours and hours trying to figure out why a perfect quantized hat pattern was going wonky. I came to the conclusion that the issue with manually recording the stems was not the Tracker itself but my setup. 

After sifting through a lot of forum posts, I found this :
https://forum.ableton.com/viewtopic.php?t=205893

I looked at my ASIO settings for my Soundcraft Mixer that I recently bought and I had the buffer rate set to 1024 samples. This was causing everything to play out of time comparable to a drunken swing. Reducing to 256 fixed this immediately. The test I was running with a perfectly quantized hat pattern was resolved. Pretty much bang on in terms of timing par a few wobbles that are inaudible.

Basically, my MIDI clock was jittering causing weird stuff to happen with timing. It's gone unnoticed with my synthesizers but the introduction of a groove box (tracker) to my setup has made it blatantly obvious. As I type this, I am manually recording stems of a recent track and will report back with the final result.

Sounds stupid but any drift when manually recording stems could be the buffer rate on your ASIO driver. It doesn't fix the song stems export issue but is a work around for me in the mean time. Hopefully, this helps some of you too :)

ancipital, 2021-01-07:
> Sounds stupid but any drift when manually recording stems could be the buffer rate on your ASIO driver. It doesn't fix the song stems export issue but is a workaround for me in the meantime. Hopefully, this helps some of you too :)

Not really relevant to this bug however. "Export Song Stems - Various Issues including tempo inconsistency". Note the use of "export".

Also, who'd manually record stems from Tracker? It suffers a high degree of SMPS noise on the analogue out at the best of times- noisier than the chorus on my old Juno :)

headtopmusic, 2021-01-07:
@ancipital If you read through the thread of this issue people are saying that there isn't a way to get any stems, whether that be exported or recording into a DAW manually. I am giving people advice on how to work around the fact the unit can't export stems at a consistent length. The unit doesn't drift on playback in general, it plays fine. That's my point.  

There's always someone who has to make a jobsworth comment. I should have known better posting on a GitHub page about Trackers that I was going to get that

doelman, 2021-01-08:
> I have contributed to this issue with my problems exporting stems. I am still suffering from the stems being inconsistent in length with strange starting points. In Ableton, the stems all mismatch in length regardless of whether warping is on or off.
> 
> I've been hesitant to post this as I am still to finish recording all of the song stems out individually since my breakthrough but thought I would share as getting the time to record them has been hard :)
> 
> After recording the stems manually and finding out the hard way that they all drift out of sync and suffer from mistiming I spent hours and hours trying to figure out why a perfect quantized hat pattern was going wonky. I came to the conclusion that the issue with manually recording the stems was not the Tracker itself but my setup.
> 
> After sifting through a lot of forum posts, I found this :
> https://forum.ableton.com/viewtopic.php?t=205893
> 
> I looked at my ASIO settings for my Soundcraft Mixer that I recently bought and I had the buffer rate set to 1024 samples. This was causing everything to play out of time comparable to a drunken swing. Reducing to 256 fixed this immediately. The test I was running with a perfectly quantized hat pattern was resolved. Pretty much bang on in terms of timing par a few wobbles that are inaudible.
> 
> Basically, my MIDI clock was jittering causing weird stuff to happen with timing. It's gone unnoticed with my synthesizers but the introduction of a groove box (tracker) to my setup has made it blatantly obvious. As I type this, I am manually recording stems of a recent track and will report back with the final result.
> 
> Sounds stupid but any drift when manually recording stems could be the buffer rate on your ASIO driver. It doesn't fix the song stems export issue but is a workaround for me in the meantime. Hopefully, this helps some of you too :)
> 
> It's not ideal but I am recording each stem with the wet signal Reverb and Delay send audible on each track so it's captured.

Thanks a lot for this analysis, definitely gonna try this (with the lower ASIO buffer) as well. Very helpful. Do you send clock from Ableton to the Tracker or do you use the Tracker's internal clock?

headtopmusic, 2021-01-08:
@doelman I am using Ableton as my clock. 'Clock in' set to USB with 'Transport in' set to USB. I recorded two tracks last night successfully with no audible timing issues or jitter. It takes about an hour but has removed the blocker of being able to make music using the tracker and send it for mastering. 

I hope this works for you. Sometimes it's the things you least expect. It was satisfying watching everything sync up as the waveforms appeared in the arrangement view of Ableton after a week of no hope.

doelman, 2021-01-08:
> @doelman I am using Ableton as my clock. 'Clock in' set to USB with 'Transport in' set to USB. I recorded two tracks last night successfully with no audible timing issues or jitter. It takes about [...]

Hero! This workaround seems to work indeed, the only difference is that I use "normal" midi-in instead of USB. Yeah it takes like an hour, depending on the length of the song but it's better than nothing

clickysteve, 2021-01-09:
> @doelman I am using Ableton as my clock. 'Clock in' set to USB with 'Transport in' set to USB. I recorded two tracks last night successfully with no audible timing issues or jitter. It takes about [...]

Just to clarify: You're having success synching with a DAW as master clock, and recording each track separately, right? If so, I might do this with the track I have sitting waiting, because I'm concerned that other updates to the FX and other features will break the track by the time the export is fixed.

headtopmusic, 2021-01-09:
Yeah, I've recorded 2 tracks using this method. It was the ASIO sample rate
knocking everything out of time. There is initially latency when the
recordings first start but I can live with that.

thebradclay, 2021-01-09:
@doelman I thought you might have found a solution for me....but, when using an external clock input on the Tracker it doesn't adhere to any Tempo changes, so I still have to manual press play to start recording and end up with ms edits. This is so damn frustrating.

thebradclay, 2021-01-09:
I think I figured out a workaround for this.....Make a blank pattern, hit play/record in ableton, hit congfig on tracker, change clock in from usb back to internal. Just tested and that works.

headtopmusic, 2021-01-09:
I haven't been using any of the Tempo FX in the Tracker so didn't come across this issue.

In that situation, I would record the stems using the internal clock and manually trim the stems. Why would you need the USB clock active in the first place?

thebradclay, 2021-01-09:
@headtopmusic Yeah will have to check later. This export issue is a total killer for me.

clickysteve, 2021-01-11:
Not sure why that quote pulled in the e-mail etc. Github. *shrugs*

headtopmusic, 2021-01-11:
Manually multi-tracking is not ideal but beats waiting 6 hours for an export for unusable material. Approx 45-60 mins per track. Hit record, then set a timer the length of the track and let it run. Come back after the alarm to trigger the next stem.

thebradclay, 2021-01-12:
I still have major drift issues with incoming stems manually. I use a macbook pro, and have tried about every configuration I can think of in ableton. This is ridiculously frustrating.

headtopmusic, 2021-01-12:
In my tests, I eliminated the Tracker as being the issue. It was my clock
jittering because of a high buffer rate. Alternatively, you could record
the stems using the Tracker's internal clock. That shouldn't drift.

headtopmusic, 2021-01-12:
To confirm, the sample rate was tweaked outside of Ableton. It was the ASIO
driver.

doelman, 2021-01-12:
> In my tests, I eliminated the Tracker as being the issue. It was my clock jittering because of a high buffer rate. Alternatively, you could record the stems using the Tracker's internal clock. That shouldn't drift.
> […](#)

Internal clock definitely drifts, I tried several times recording the separate channels but it got all messed up. Using Ableton as a clock with a low sample rate is doable to record stems for me.

thebradclay, 2021-01-12:
I don't use asiorall as I am using a mac. I use tempo changes in the tracker in pretty much every song which makes recording stems manually a pita. I just sat for 3 hours recording stems. 51 slots 31 patterns. drift is in random places and it's really impossible to line things up.  I'm using the Ableton clock with the lowest possible sample rate and it's still fucked. I'm probably going to sell this thing because I can't actually get anything usable out of it which sucks because the workflow is fantastic but this may be the last straw as I have this project due on the 15th for a charity streaming event and now pretty much have to do two weeks worth of work over.

[1 more reply]

closed 2021-01-15 by shuler323