polyend/TrackerBetaTesting #514: 1.3.0.b3] Stem render issues/Sample Library inaccessible after rendering stems

thebradclay, 2020-12-13:
### Render a songs stems then hit sample loader

Render a songs stems

hit sample loader

Under SD card nothing is there

After rendering stems there is no audio and Ableton cannot read or load any of the files.

### Attachments and links

[image]

### Other important information

* How often does this bug happen? I have rendered the stems twice and this has happened both times.

* Does this bug happen in previous firmware versions? No, however the song that was rendered was made under the previous beta fw if that matters.

* Does this bug happen in every project? Yes

shuler323, 2020-12-16:
@thebradclay please attach project which generates this bug, thanks

thebradclay, 2020-12-16:
Hi Shuler323,
Project attached below:

[12122020TEST.zip](https://github.com/polyend/TrackerIssues/files/5704187/12122020TEST.zip)

shuler323, 2021-01-15:
thread moved to #697

closed 2021-01-15 by shuler323