polyend/TrackerBetaTesting #516: [1.3.0b3] Folders with 1000+ files not loading

secretkiillerofnames, 2020-12-14:
I have some folders with 1000+ files that are divided out into multiple folders as per the 1.2 update.
I've not had problems with them before.

On 1.3.0b3 I get a freeze whenever trying to access the folder via the Sample Loader.
I rolled back to 1.3.0b2 and the problem does not occur.

### Steps to reproduce

Click to access a split/divided folder via the sample loader in 1.3.0b3
Instant freeze

### Attachments and links

A video of the freeze.
https://www.dropbox.com/s/kn2rt505ovb59qe/130b3folderproblem.MOV?dl=0

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
All the time with 1.3.0b3
Not at all on 1.3.0b2

* Does this bug happen in previous firmware versions?

Not in my experience.

* Does this bug happen in every project?

I've had it happen in a new blank project and a pre-existing one.   Haven't tried ALL my projects.

cheers
Lloyd

secretkiillerofnames, 2020-12-14:
Did a bit more testing.  
I have a folder with 3 separate divisions that works fine in 1.3.0b3
But one with 6 divisions, and any above that, cause the freeze.

discohead, 2020-12-21:
I’m also experiencing this exact issue in 1.3.0b3. Likewise, was not a problem in previous firmwares for me.

secretkiillerofnames, 2020-12-21:
I've tried breaking up the files into small subfolders and it is still freezing even with only 2-300 files per folder.

secretkiillerofnames, 2020-12-22:
Seems to be fixed in b4 and b5.

closed 2021-01-04 by ambv