polyend/TrackerBetaTesting #518: Change pattern numbers based on their order in song mode

mroeLGC, 2020-12-14:
When writing music on the Tracker I often go back and add new sections to previous parts of the song. This often means that I will have the intro section of the song as a relatively high pattern number compared to the section directly after the intro. I then have to  Page + scroll backwards and forwards a lot to edit these adjacent sections. It also means that it's harder to keep a mental model of what patterns are related.

It would be great if there was an option in Song mode to rearrange the patterns so that their pattern numbers correspond to the order you put them in while in Song mode. It would make songs with a high pattern count easier to work with.

ketumatek, 2020-12-14:
I ask in an older post if we can mane slot in song mode...as this it is easier to know where you are and what slot use...
They can make some short name that may be we can choose or custom as

Intro 
Break 
Thème 
Bridge 
End 

And more if you have idea..

benc-uk, 2020-12-19:
slots or sections would make it easier to navigate large and complex songs