polyend/TrackerBetaTesting #522: FILE TRANSFER MODE - MacOS

otternet, 2020-12-15:
Hi Polyend, I’ve tried every app available for a Mac to use MTP, but somehow it just doesn’t connect. Maybe because I am using a Mac with USB-C only and using a dongle or hub, SyncMate, Commander One etc does not recognize PT when using the File Transfer mode. Some claim to have success with it, but I can’t seem to. Can you find a protocol that’s agnostic and allows Macs and WinTel machines to use this file transfer feature?

jgrimsong, 2020-12-15:
I had same problem until I used Android File Transfer tool. That's the only one that worked for me.

otternet, 2020-12-15:
> I had same problem until I used Android File Transfer tool. That's the only one that worked for me.

How did you connect it to your Mac? Can you share your steps? I downloaded the same, but it doesn't work for me. Are you running Big Sur? I've even tried OpenMTP but nothing seems to work.

jgrimsong, 2020-12-15:
I just installed it on my iMac and have the USB cable attached to my hub. I'm running Catalina. Everything just came up automatically.

otternet, 2020-12-15:
I'm using a dongle to my MacBook Pro running Big Sur but nothing shows up  🤷🏻‍♂️ Gonna try that again.

otternet, 2020-12-15:
[image]

Nope... grr! Still doesn't work. Maybe because I'm using the USB-C dongle to my MacBook.

b0ll3, 2020-12-16:
Let's start with the most simple thing first:
is your device in File Transfer Mode?
Tracker's screen should look like in the picture.
[image]

otternet, 2020-12-16:
Yup it is. I've also tried different "startup" sequences. Launch app first and then switch on FTM, and vice versa. Used a hub, used a dongle. The only thing I have not tried is plugging the included USB cable from Polyend to USB-A input, but my MacBook Pro only has USB-C. I have the "first generation" Tracker (flat keys), and from what I understand the second generation Trackers have a more flex USB-C to USB-C connection. Am wondering if that's the problem here.

otternet, 2020-12-16:
UPDATE: no luck with direct to USB-A input sans hub or dongle either 🤷🏻‍♂️

Polyend... any suggestions from y'all?