polyend/TrackerBetaTesting #523: [1.3.0b3] Crash while accessing Master page

phaelam, 2020-12-15:
so I had reported a bug I couldn't reproduce, until now.

what happens is while im in Rec mode, after ive placed some notes for Intrument samples, I go into the Master page to adjust Reverb, and I got a crash. I cant even shut off the PT.

I believe these are the same steps that caused notes to change in the patterns.

piotr3500, 2020-12-16:
Thanks for reporting it. 

Does it happen every time? 
Does it happen on new projects too?

Is it possible for you to send a short video of the steps you are making to reproduce the issue?

phaelam, 2020-12-16:
> Thanks for reporting it.
> 
> Does it happen every time?
> Does it happen on new projects too?
> 
> Is it possible for you to send a short video of the steps you are making to reproduce the issue?

trying to reproduce. I cant remember my exact steps.

I start with a new project. one crash happened when I inadvertent held shift as I went from the empty samples slots to the sample list....and was stuck in a cursor mode. all the field would turn red as I moved up and down the list, then it crashed.
another time was when, in Rec mode, after copy pasting a bunch of stuff, I went into Master page and it hung, hard crash.  power button didn't werk and I had to unplug to power cycle
last time was while in Rec mode, copy pasting a bunch of CC trigs, went to master page and hard crashed, unplug to power cycle.

one consistent thing I'm noticing is it happens while the pattern is playing, im in Rec mode, and I enter the Master page. as soon as I hit the button, it hung. 

again, trying to get the steps to reproduce. once I do, ill video it. im wondering if the CPU gets a little mixed up when doing a lot of copy pasting. just a thought. I have audio of the crash if you want....not that that will really help.

closed 2022-02-06 by MortalJS