polyend/TrackerBetaTesting #525: Metronome use in more possibility

ketumatek, 2020-12-15:
(1.3.0.b2)
At this moment I think I can use metronome only in recording mode...that is great but not enough.

It should be super helpful if we can use it in play mode...
Why? Because before recording you have to be sure of what you want...
For example I don t have any sound in the first pattern. I take a bass sound and I want to play with it ...if I have the metronome I can have a reference to play with in tempo..
So when I recording after I known exactly how my phrase would be...
Without it is very difficult and a waist of time.

So may be you can put 2 options like

1- Play mode 
2 - Record mode 

Hope it make sense

jaap3, 2020-12-27:
Isn't this pretty much the same feature request as #478?

ketumatek, 2020-12-27:
> Isn't this pretty much the same feature request as #478?

Sorry if I put it twice...may be I didn't remember that I post it before