polyend/TrackerBetaTesting #526: Gate Length

phaelam, 2020-12-16:
small request but it would be nice to be able to set the gate length of a step. currently the base setting ends up being too long. i DID find a werk around. if you have two steps, one after another, [im sending gate to modular] the first step overrides the following step. I tried placing the two triggers on separate tracks, but the first step is just too long. so to fix that I put a roll of 1, R1 on the step and that allows for the following step to cut off the first step before the second gate of the Roll can fire.

but setting gate length might be useful. and you could use Randomize FX on it for some interesting results.

[if there are ways to do this already....please lemme know!]

infamedavid, 2020-12-18:
+1 and I want to suggest,  it will be  great as fx , not as a global setup

phaelam, 2020-12-18:
Ya I meant to request it as an FX lane option so you could use a Randomize FX on it to get varying gate lengths

LetsSeeIfWeSolveThis, 2020-12-18:
Already requested #265

phaelam, 2020-12-18:
I’m sorry...I guess I asked the wrong question in the search :(