polyend/TrackerBetaTesting #534: [1.303b] Audio crackle in line In monitor when many tracks are being used

prds100, 2020-12-19:
Hi

Line in live monitoring crackles when there is a lot of step information in the pattern.  Currently running 5 tracks out of 8 and the audio crackle is now present when playing the monitor in (level is not hot)

discohead, 2020-12-21:
I've experienced this as well, though I didn't have a lot of step information... maybe only 2 or 3 tracks. I thought it might be the beta firmware, so I re-installed 1.2.2, realized it doesn't have line-in monitoring, re-installed 1.3.0b3 and the issue was gone, same project, same signal at the line in, just swapped the firmware out and back. Maybe a memory leak somewhere?

EridonHermetz, 2020-12-21:
i also have the same crackle in the Line In with the 1.3.0b3

and it seems to appear when you are manipulating the limiter section (on the Master Page) 

i'm sure that this is a memory leak somewhere like you said

piotr3500, 2020-12-22:
fixed in b5

closed 2020-12-22 by piotr3500