polyend/TrackerBetaTesting #535: Song Mode - Visual feedback and usability

benc-uk, 2020-12-19:
I feel song mode could do with better visual feedback indicating differences in patterns
- Colour code or otherwise indicate different patterns, the pattern number on the far right is the only clue in an otherwise sea of identical white rectangles that there might be any difference between two slots. Simply cycling with modulo arithmetic through a set list of 7 or 8 colour values for the patterns numbers would suffice.
- Pattern length is not shown anywhere, I'll often copy a block from one pattern to another without realising the patterns are different lengths. Rather than a number, some of visual indicator like a bar length or shading could be used

ketumatek, 2020-12-19:
I ask I bit same question before ...colors per tracks...and mane for slot...
I m agree with you, not really easy to recognize quickly the different patterns and what is inside just because everything is the same and ou have just a number...
Sure they can make a list like . intro . Break .thème .etc and custom word..

eimhin, 2020-12-19:
Good idea. It would be helpful if the track rectangles had opacity/saturation/brightness change depending on the amount of note/fx activity going on in them. 

I find the on-or-off state of them right now doesn’t give you a clear picture of what’s going on

rotarycuff, 2021-02-08:
It would be great to be able to name the cells in the song mode, so it becomes easier to remember/see what is what when copy pasting.

The tracknaming on the top is already something, but just being able to name the cells by number for example, would make the overall overview even better. (Even if it would only allow 2 or 3 digits to fit within the cell)

(a bit similar like in nanoloop where its just numbers linking to a part)