polyend/TrackerBetaTesting #539: Sample Editor improvements. Zero Crossing indication + Cursor Snapping + Zoom Levels + Y Axis Zoom

meteorist, 2020-12-20:
I would love if the sample editor was improved with these features.

Zero crossing on/off
When you loop a sample the starting point of the sample needs to be relatively close in amplitude to the end point of the loop so that a clicking artifact is avoided. This can be done by fiddling around with the start and end cursor and sometimes you're lucky and somethimes you're not. A helpful tool would be a Zero Crossing indicator where the amplitude is the same across the sample. This doesn't necessary need to at zero level but could also be a set value. Setting this value could be done by taking the sample amplitude from where the last cursor moved is located. 

Snapping on/off
Enabling a snapping mode would allow for quicker jumping between the zero crossing points and make it easier to find good loop points.

Zoom levels
I also request deeper zoom levels and the addition of Y axis zoom.

Colors
The zero crossings would be highlighted with an accent color.
It would also be nice if the zero line would be gray instead of the same white color as the sample points/bezier line.

benc-uk, 2020-12-21:
Holding shift while moving the wheel to snap to zero crossings just like on the Octatrack makes sense.
In fact I immediately tried this in the sample editor and was surprised it was absent

tfspsound, 2020-12-21:
Even something simple like a fine line that runs from the start point to the end point that could act like a spirit level ie turn green when it was level. That be cool.

keith-macleod, 2020-12-22:
Being able to zoom into single sample scale and even slightly beyond is also very useful for accurately editing and looping. The sample edit/loop/slice views should really be showing position in samples, too. Seconds are neither particularly relevant nor precise in this context.

drowsy-net, 2021-01-30:
Fantastic ideas. I hope these are added!!