polyend/TrackerBetaTesting #542: [1.2.1] Slices unaffected by Sample Length Performance Effect

tfspsound, 2020-12-21:
Steps to reproduce

I made a beat and used a Sliced sample on one of the tracks, the rest were one shot samples. Then I applied Sample Length to the track in Perform mode. Sample length adjusted all the one shot samples and ignored the sliced samples. Not sure if it was Beat Sliced or just Sliced not at my tracker now. Other important information

* How often does this bug happen? every time

* Does this bug happen in previous firmware versions? 

1.2.1 

* Does this bug happen in every project?

yes

ambv, 2021-01-05:
Sample Length literally affects the entire sample's length, not slices. We don't consider the current behavior a bug at this point.

closed 2021-01-05 by ambv

tfspsound, 2021-01-05:
Hi
I wasn’t sure if it was a bug at the time but it would be helpful if it were to include the individual clip lengths to create a chopped / gated sound. I think sample length sounds great on beats especially, and there is a beat slice option so how about, if possible, you make sample length apply to that. I use slice option more than individual samples because they seem more efficient and it’s so much quicker / inspirational to have access to multiple sounds at once. My other point would be that if a user doesn’t want to effect a sliced sample they could always disarm the track. 
Cheers for reading these your work is appreciated.