polyend/TrackerBetaTesting #547: Synthesis improvements

keith-macleod, 2020-12-21:
I find it difficult to impossible to get usable results from either the granular or wavetable engines due to popping and tearing artifacts, particularly when modulating the position. A couple of things stand out to me:

The granular engine doesn't seem to incorporate any kind of window blending which typically means that the 'square' window produces the smoothest result (after tweaking the window to avoid popping artifacts,) since it doesn't have the effect of amplitude-modulating the waveform. That smoothness disappears however, when the position is modulated, since the careful window tweaking only applied to that one position.

What one would expect with the triangular and gaussian windows is that they would be evaluated in such a way that the current window and the next window are summed so the total contribution of both sums to 1, producing a smoothly interpolated result. However, from what I can tell, no summing or interpolating is taking place, resulting in a full-level amplitude modulation at the frequency of 1/window size.

So request 1 would be to have window interpolation/summing in the granular engine.

For the wavetable engine, I've not had much success creating my own wavetables or importing wavetables from other sources. The WT smoother doesn't seem to help much here. A more fleshed-out set of tools for stretching and aligning waveforms to zero-crossings would go a long way here. I'm still not sure why the WT smoother offers both a window length and window number options when the selected values would be clearly longer than the sample. I also don't understand why it would offer window lengths that the playback engine doesn't support.

Another thing I've found in my sound design adventures is that the display of short waveforms can be buggy, perhaps because of zoom quantization limitations, but frequently when trying to work with a single cycle waveform, I can't set the endpoint because the editor can't zoom to a scale where the sample occupies less than the full view, but the next available zoom size is such that the end of the wave is cut off.

tfspsound, 2020-12-22:
Attack and decay on the grain window would be nice, I don’t find the tearing particularly musical either. It’d be nice to have an option for more grains but I’m don’t know if the tracker could handle it.

MetabaronCZ, 2021-01-23:
I also played with wavetables lately and the popping sound makes it useless. I can use instrument volume envelope to get rid of some of the artiffacts, but then I cant use the envelope to modify the sound.

I think that every note played should have about 10ms fadein on NOTE ON and other 10ms fadeout on NOTE OFF / CUT by default, so no popping ever occurs. Or it could be some boolean parameter of the instrument.

Also, turning off sound anti-aliasing in project settings makes some wavetables dramatically better.

doelman, 2021-01-23:
Yes indeed, this popping issue is quite hard to work around. Not only with the wavetables, also with the LFO panning for example. That functionality sometimes introduces an extreme amount of popping /clicks.

MetabaronCZ, 2021-01-24:
Oh yes, panning LFO is also useless with a lot of samples.

mwmwmw, 2021-02-01:
I have some options/our of the box ideas. 

**Prerendering**
When rendering the granular synth, provide an option to increase the number of grains. The granular synth becomes a sound design tool with that old school, pre-rendered flavour. 

**Track stealing**
Provide an option to dedicate more tracks or effect channels to the granular synth. 

**4-track Mode**
A 4-channel 'HD' mode with interpolated value changes on pan and position. Glide seems smooth.

**Shut off Reverb**
The clouds reverb is sort of a granular synth as well (https://github.com/pichenettes/eurorack/blob/master/clouds/dsp/granular_processor.cc). 

I'd rather have no clicks than reverb, every day, forever.