polyend/TrackerBetaTesting #548: Line In Toggle Mono or Stereo

bichan17, 2020-12-22:
It would be great to be able to toggle the Line In input as Mono or Stereo in the settings, so I can play my mono instruments thru the tracker. Currently when I try to run my guitar on this input I can only hear it in one ear and there is no way to pan the line in signal.

sunfallz, 2021-01-25:
Definitely second this! 
In the manual it states “use a mono or stereo cable”, but:
Currently, when sampling my modular into the tracker, using a mono patch cable, i can only hear the left side out of the tracker output as well.
Example: Sampling bass to make kick drums does not require a stereo signal. 

Using a Mono cable to sample should normalize  to both left and right when monitoring output.

And/or have the option to quickly toggle mono or stereo in the sample recorder page.

wraggal, 2021-01-29:
yes please!

Jacknametrouble, 2021-12-02:
done

closed 2021-12-02 by Jacknametrouble