polyend/TrackerBetaTesting #549: [1.3.0b4] Starting not Starting

db9979, 2020-12-22:
## Hi, sorry you had trouble!

### Steps to reproduce
If I push Play only the time in the upper bar starts and counts the seconds up but the track/pattern won't start. 

### Other important information

* How often does this bug happen?
Every time on all projects. 

* Does this bug happen in previous FW
In 1.2.2 this does not happen

* Does this bug happen in every project?
Yes

dimant, 2020-12-22:
I can confirm this reproes for me too. Works fine with 1.3.0b1

math0ne, 2020-12-22:
Same behavior here on 1.3.0b4

isbtn01, 2020-12-22:
Same here.
Thank you.

Timour, 2020-12-22:
Confirm same here on 1..0b4. If I switch to midi clock usb and back to internal clock, while playing, it starts playing. So seems like a clock problem.

piotr3500, 2020-12-22:
It's fixed. 
Please download and install 1.3.0b4 once again.
Sorry for that!

closed 2020-12-22 by db9979