polyend/TrackerBetaTesting #551: [1.3.0b1] Beat slice - always plays last slice

keith-macleod, 2020-12-22:
Hi, I've just run into this problem, I can't be sure if it coincided with the latest update, but certainly not before.

When I use a sample in beat slice mode, the last slice is always played. It seems to happen both when triggered by pad and when triggered by the sequencer.

This happens with all projects.

I've tried downgrading the firmware as far as 1.3.0b1 and the problem persists. I also tried using my old SD card, which I haven't touched since 1.1, thinking maybe there was some sort of corruption in the workspace directory, but there was no change

alexbuga, 2020-12-22:
Fixed. You can close it.

closed 2020-12-22 by piotr3500

LetsSeeIfWeSolveThis, 2020-12-22:
> Fixed. You can close it.

It's fixed on b4 for you? I still have it.