polyend/TrackerBetaTesting #552: [1.3.0b3] more filter types

jtkthx, 2020-12-22:
Simple request:  In the list of filter types, options for 2-pole and 4-pole low pass and high pass filters.

Even better would be modeled filters like moog or oberheim, but I know that's a lot to ask.

Displeak, 2020-12-22:
I like this idea ! I'm pretty sure there are some nice open source filters from the eurorack world

bambrose, 2020-12-24:
Yeah the preen2FM open source project has some nice filters (Bass is a good one)

fragment23, 2020-12-29:
+1 Notch and two peaks (where the peaks could be moved in spacing to each other) would be great

danapeks, 2021-01-06:
Adding my name to this. Not hugely loving the sound on the existing filter, a few more options would be ace!

bambrose, 2021-01-06:
FWIW I love the existing filter. One of the favourite digital filters I've ever used .. so don't change it !! just some more options would be cool if possible

static-noise-PT, 2021-04-20:
A would prefer to first see a comb filter (would step up sound design a lot) and than on second different slopes and maybe a notch one.

closed 2021-12-02 by Jacknametrouble