polyend/TrackerBetaTesting #553: general audio latency

bnckd, 2020-12-22:
Hey there, 
i'm having this overall latency in tracker (i think it's 15 ms) which is pretty annoying. 
not sure this is a tracker issue in general or a specific issue of my unit, but basically i's not possible to have the tracker playing alongside a drummachine or triggering modular drums via midi cv on the side. 
this happens with PT as master and as slave, with internal, midi & usb sync. 
i've tried numerous samples which are precisely cut, various drum machines and setups, literally everything but the tracker runs tight, no matter what. all the machines go directly into a mixing desk, so this isn't about interface latency or so. 

attached find a screenshot, the third track is the PT -  
also attached is a short audio file with first the 606 plus mod then the tracker added, playing the same rhythm 
pt is the master but changing that won't change the latency at all. 

[image]

[https://www.dropbox.com/s/dmgxn27ygrncxk9/pt%20latency.mp3?dl=0]

piotr3500, 2020-12-22:
did you try 1.3.0b5 firmware and Clock sync delay option?

bnckd, 2020-12-22:
hi piotr, yes of course, i was waiting for that one.
so first - internal doesn't work anymore with 1.3.0b4 - maybe it does with
the newer one.
apart from that the fix is only delaying incoming midi clock, which to my
understanding would not affect the latency of the device itself, the audio
latency.
it's super weird and i mean if you try to reproduce that scenario - does it
sound like in my link? and look like that?
thanks for the quick answer!

bnckd, 2020-12-22:
hey piotr. so well, now also the midi out is wonky as hell. i was happily triggering my whole modular rack plus clocking drummachines with the tracker - without tracker audio unfortunately due to latency. now after the last beta also the midi is all over the place. sometimes it just has irregularities during a 32step sequence and with a 16step sequence the tracker seems to make tiny breaks at the end of each cycle, so everything else clocked is running out of time plus sounds triggered via midi are like 30ms off. reverted to official 1.2 and the glitches are gone but the same sequence won't trigger anything. this is really messy ;) back to the new beta and this is what the glitches in clock sound like: 
https://www.dropbox.com/s/1hwowzkgerzv2v3/glitchclock.mp3?dl=0

ambv, 2020-12-22:
@bnckd we would like to get to the bottom of your issue because what you're reporting goes against what other people are seeing (which is that 1.3.0b5 is an improvement in both clock stability and that the latency compensation works).

I'd like to know more about your setup:
- what is clocking what
- what is the role of Ableton Live and how are you connecting Tracker's out to your computer
- what is your Ableton buffer size and reported overall latency inherent to your setup? I mean this information:

[image]

We need to know more about what you're doing to be able to diagnose the problem you're experiencing.

bnckd, 2020-12-23:
hey there, great, yes. 

so - 1) different clocking scenarios. usually i work otb, so usually tracker is the master and clocking various other midi devices as well as a midi cv clock. when recording, i switch to usb sync and slave the tracker to ableton. 

2) as said, ableton is only there to record. in fact i (can and sometimes do) run drum vsts or precise delays etc via my mixing desk and have less latency than the tracker audio. i'm not monitoring with ableton (only during recording). my setup is more like a live situation: 32 channel mixing desk with lots of inputs, aux sends and all the usual stuff. i then route whatever goes into ableton, record live and go from there. 

3) latency sceenshot coming up! 

[image]

bnckd, 2020-12-23:
ps: if you've heard the two audio examples you know i'm suffering ;) and i really want to make this work as i need the tracker as my main hub. it clocks a rd6 and a 707 via midi through hub plus sends midi data to the squarp rample and 2 synths, as well as midi cv to my modular system.
haven't used its audio except for standalone sketching so far due to above mentioned reasons.

bnckd, 2020-12-23:
it works! glitches are gone, the audio latency is gone, heureka! you guys definitely rock, banging. nice holidays and keep it coming ;)

discohead, 2020-12-24:
@bnckd this issue is fixed for you in 1.3.0b6? I'm still having issues myself: https://github.com/polyend/TrackerIssues/issues/570

bnckd, 2020-12-24:
hey, yup it solved it so far, had no latency for the first time since august ;) 
i tried it only with tracker slaved / synced via usb and about -10 ms i think, but i'll check later again, just to be sure. 
i used the same sounds and same chain as yesterday, when things got really bad with midi & audio. 

i'm not sure (haven't tried last night) if this is also gone with internal sync - will try it in an hour or 2. 
but that would be very important as well - as a lot of folks use it as the central brain (same here..)

bnckd, 2020-12-24:
so tried it with the new beta and the new firmware, clock is more or less stable - it's not 100% tight but 98 % ;) 
and also works when clocked internally!

closed 2020-12-24 by bnckd