polyend/TrackerBetaTesting #554: [1.3.0b5] Glitch in Instruments Parameter

EridonHermetz, 2020-12-22:
## Hi, sorry you had trouble!

### Steps to reproduce

Pick any instruments

Go to the Main Parameters Instruments

Choose the Enveloppe Parameter in any mode (wavetable lfo , volume , etc etc)

Change the "Amount" Value , and ... it is still moving while you are changing AN OTHER parameter

It is locked somehow in the "Amount" parameter whatever you are doing once you selected it

The only way to "escape" the bug is to click again on "Instruments Parameter"

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Every Time

* Does this bug happen in previous firmware versions?
Nope

* Does this bug happen in every project?
YES

piotr3500, 2020-12-22:
Can you check if something similar happens when you change Cutoff?

EridonHermetz, 2020-12-22:
Yes , it happens too

Every parameter , on the Automation Page (2/2)

Every time you select the Amount , you cant deactivate it and move whatever you are choosing

piotr3500, 2020-12-22:
Got it, we're on it.
Thanks for reporting.

closed 2020-12-23 by ambv

EridonHermetz, 2020-12-23:
there is something to download to have this fixed ?? or should i just redownload the firmware ?