polyend/TrackerBetaTesting #555: [1.3.0b5] Button functions get corrupted

mkraemer, 2020-12-22:
All of the following is reproducible and started after flashing 1.3.0b5:

### Example 1: Sample Loader button starts opening Sample Recorder screen:

After power cycling the tracker, initially the Sample Loader screen can be opened as expected.
After loading any song (e.g. the included demos), the Sample Loader screen can still be opened.
After playing any song, the Sample Loader button now opens the Sample Recorder screen; the Sample Loader screen can't be reached at all now.

### Example 2: "More" screen key opens Render Selection and Fill screens

After power cycling the tracker, the "More" screen key gives access to the other 8 functions.
After pressing the record button, the "More" screen key now opens the Render Selection screen.
When cancelling the Render Selection screen and pressing the "More" screen key once more, the Fill screen is shown.
When cancelling the Fill screen and pressing the "More" screen, the Render selection screen is shown.
And so on..

### There's more strange stuff happening:

Assuming the cursor (green box) is at step 1 of track 1, pressing record doesn't simply turns the cursor red. The cursor does get red, but the selection is now widened to step 1 and 2 of track 1 and 2. Not sure if related as it's not a complete corruption (as recording mode _is_ triggered), but it started at the same time. So it might be related? :man_shrugging: 

I tried downgrading to 1.3.0b3, which now shows the same behaviour.
Downgrading to 1.2.2 breaks all functionality. I can enter the config menu to reset the config and I'm able to flash another firmware, but pretty much anything else (playing a sample, pressing the file button, ..) freezes the tracker completely.

piotr3500, 2020-12-22:
We have non of your strange behaviour.
The issues seem to be related to hardware, not software.
Please contact us via polyend.com/help so we can resolve it asap.

piotr3500, 2020-12-22:
For the last shot, please try formatting SD card - perhaps it is broken.

closed 2020-12-22 by piotr3500

mkraemer, 2020-12-22:
> For the last shot, please try formatting SD card - perhaps it is broken.

Didn't help, unfortunately. Thx, will contact support.