polyend/TrackerBetaTesting #560: Different length for tracks

60pfennig, 2020-12-23:
It might be not the oldschool tracker style, but why not offer some more possibilties. I think it would be nice to assign different track lengths per pattern and as far as I've read there are many more people wishing the same. So maybe it will get possible someday  :)

tilllt, 2020-12-24:
That would enable some polyrythmic madness

Tombelliard, 2021-01-21:
I agree !!

gmli, 2021-01-24:
The Tracker is already doing it in performance mode, so enabling it in pattern mode should be possible. The only thing to visually change is the playing « cursor », maybe it should follow the one on the currently selected track.

sunfallz, 2021-02-04:
Wonderful suggestion 👍

sunfallz, 2021-02-04:
> The Tracker is already doing it in performance mode, so enabling it in pattern mode should be possible. The only thing to visually change is the playing « cursor », maybe it should follow the one on the currently selected track.

side note: When using the pattern length fx in performance mode, you're effecting every track at the same time, not individual tracks, essentially you can only have "pattern length" as an effect slot on 1 track at the moment.
so great feature request 💯

closed 2021-12-02 by Jacknametrouble