polyend/TrackerBetaTesting #568: Clone Pattern to new Slot in Song Mode

tilllt, 2020-12-23:
Would be great if song mode had an option to duplicate a pattern to a new Slot easily. I have seen this issue
https://github.com/polyend/TrackerIssues/issues/458

Since I am still using the stable release, I don't know if the "duplicate pattern" mentioned in the issue above is referring to my feature request, if so, feel free to close it.

bgdnrz, 2020-12-26:
Aside from the method to access this functionality, how does it differ from 'Copy Pattern' in Pattern mode (Pattern mode > More > Copy Pattern > move to new pattern > Paste Pattern)?

tilllt, 2020-12-26:
It does not, it would just reduce several button taps to one button tap. This process is probably THE most important step in constructing a song, so it makes a pretty big difference if it is:
- Copy Pattern
- Move to new Pattern
- Paste Pattern
- Go to Song Mode
- Assign Slot to New Pattern

Or just (in Song Mode)
- Clone Pattern

Let's say you are using 100 individual patters in your song, that's 500 button taps vs. 100.

bgdnrz, 2020-12-26:
I like the sound of this workflow. I have been doing the same function via pattern mode and it's a bit cumbersome. Shift + Add Slot from Song mode sounds like a more intuitive and quicker method!

bn-mk, 2021-01-07:
I think this would really help the song mode - I'd also like any changes i make to a pattern in song mode to only affect the instance of the pattern i am editing, rather than every instance of that pattern - eg. deleting a track section from pattern 1 should only affect that single instance of pattern 1 and not any other pattern 1's i may have in the song.

I guess that would mean cloning the pattern in place and auto-assigning a free track number each a slot is edited

ketumatek, 2021-01-27:
I confirm that, in song mode, it is an easier and quicker way to arrange your song, if you can copy directly each pattern (slot) you want and Paiste them in a empty new slot.
More than create a new pattern copy what you want from the other pattern and paiste them in...
Way more long...

ketumatek, 2021-01-28:
> I think this would really help the song mode - I'd also like any changes i make to a pattern in song mode to only affect the instance of the pattern i am editing, rather than every instance of that [...]

Completly agree with you ...each pattern in song mode should be independent from the first to the last one (255 if I remember).
Because right now if you change one thing in a pattern you need to create a new one if you don t want to affect the other...

At the moment the song mode it is like what you have in a elektron Machine Drum with number you can choose and put when you want.
And what we would like it is more how Renoise work..lines and you can put what you want in. You can copy and paste which pattern you like to put in the track and the slot or line you want..

ketumatek, 2021-01-28:
> It does not, it would just reduce several button taps to one button tap. This process is probably THE most important step in constructing a song, so it makes a pretty big difference if it is:
> 
> - Copy Pattern
> 
> - Move to new Pattern
> 
> - Paste Pattern
> 
> - Go to Song Mode
> 
> - Assign Slot to New Pattern
> 
> 
> 
> Or just (in Song Mode)
> 
> - Clone Pattern
> 
> 
> 
> Let's say you are using 100 individual patters in your song, that's 500 button taps vs. 100.
> 
> yes plus sélect multiple pattern copy and paiste in new slot.
And add total empty slot.

ketumatek, 2021-01-28:
> Aside from the method to access this functionality, how does it differ from 'Copy Pattern' in Pattern mode (Pattern mode > More > Copy Pattern > move to new pattern > Paste Pattern)?

Waist of time and unpractical way for song mode.
For me more easy to copy and paste the pattern you want in the new slot...directly from song mode..

jeryanders, 2021-06-26:
While I was composing in song mode recently, I had feedback similar to this issue. This would be an excellent feature.