polyend/TrackerBetaTesting #570: [1.3.0b6] Internal Sample Latency vs MIDI Latency

discohead, 2020-12-24:
### Steps to reproduce

1. Play an internal sample and a MIDI note on the same step.
2. Record both outputs into DAW
3. Note that the internal sample is always recorded ~20ms after the MIDI triggered sound

Please describe the steps to reproduce the bug. Tell us what you noticed
happening and what you expected to happen instead.

I'm just triggering a sample and MIDI instrument on the same step and recording the output of the Tracker and the MIDI instrument on separate channels in Ableton. Tracker and Ableton are both slaved to an external clock from the Expert Sleepers FH-2. Using the MIDI delay I can get shift the two signals around and I can get one or the other on the grid, but the sample is always about 20ms later than the external MIDI instrument.

### Attachments and links

[image]

[image]

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
every time
* Does this bug happen in previous firmware versions?
unsure
* Does this bug happen in every project?
yes

discohead, 2020-12-24:
The external MIDI instrument and Tracker are both being recorded via my UA Apollo interface. As mentioned, master clock is an Expert Sleepers FH-2, both Tracker and Ableton are slaved to that clock. The Tracker MIDI note is converted to a trigger output on the FH-2 which is being used to gate some noise that goes directly to my audio interface. The Tracker output is also connected directly to my audio interface. Ableton live latency settings are...
[image]

ambv, 2020-12-24:
Did you try using a setting like -40.0 for latency compensation on Tracker?

discohead, 2020-12-24:
@ambv Yes, but the Midi clock sync delay ultimately has no effect on this problem since the delay is applied equally to both the internal instrument and the midi instrument. Adjusting the delay shifts both events forward/backward in time but the internal sample is always recorded ~15ms-20ms late, relative to the midi instrument. 

Are you not able to reproduce the issue?

joe-chip-cyber, 2021-01-04:
I am able to reproduce this issue easily using firmware v1.3.0, as well as each of the 1.3 beta releases.

Tracker audio output is always 15-20ms behind the audio coming from external midi gear that is being sent Note information from the Tracker.

This seems to be a latency issue with the Tracker audio output, and this can be shown by using micro-moves to delay the midi notes being sent to the outboard gear.

At a project tempo of 120, a micro-move value of ~15 on the Midi notes brings the Tracker audio and external gear audio output closer in sync.

ambv, 2021-01-06:
@discohead as your image shows, your roundtrip audio latency is 17.8ms. Try shortening your audio buffer in Ableton Live and you'll see the audio latency decrease.

As for internal audio generation being late vs sent MIDI, it's a consequence of having audio buffered. We'll see what we can do about lining those up.

ambv, 2021-01-06:
@joe-chip-cyber is onto something with the MIDI micro-move lining things back up. We should maybe do this automatically.

ambv, 2021-01-06:
@discohead, @joe-chip-cyber: in the mean time in Ableton Live you can compensate this on a per-track basis manually:
[image]

joe-chip-cyber, 2021-01-07:
Nice suggestion for per-track delay in Live, that's a clean way around. 

Unfortunately Logic Pro X only allows this delay as a "region" setting for audio tracks (per-track _is_ available on MIDI tracks, but this doesn't help me out, since I'm only dealing with the audio coming from Tracker and the other sources). The region delay does allow me to delay the playback of already-recorded audio, but the problem remains for live monitoring of the sound.

Thanks for checking this out @ambv! Look forward to any updates

AlexNorinh, 2021-01-14:
I appreciate this issue has been acknowledged as consequence of audio buffering; just posting to confirm I also experience sampled audio v midi latency differential

(1.3.0, digitone clock-slaved to tracker via MIDI. Internal or external midi sequences in sync, samples consistently off, no drift. I had the audio from digitone routed into tracker, so I was suspicious there may have been some audio processing lag going through the device, but when routed the other direction, yields the same result).

NoraCodes, 2021-01-15:
I'm curious: how does Line In passthrough interact with this? Is that audio buffered too? If so, please ensure the fix doesn't cause that opposite issue for the use-case of the Tracker as a middlebox between a synth being controlled over MIDI!

bnckd, 2021-01-27:
I'm having the same issue. The workaround here is the audio latency setting and not using the tracker for midi sequencing anymore, which is a pity, as I always need the mac plus ABL live on. But my real problem is that I would like to use it detached from the computer as a main hub and this is not working out at all after almost half a year, as audio still is always late. Kind of a bummer and I hope they'll do something about it, I wrote them several mails and also addressed this issue in a bug report regarding the external sync latency, but there has never been a word about internal sync audio latency. I wonder why nobody addresses that, it's not tight at all.

bnckd, 2021-01-27:
Ps: if it's the audio buffer I suppose this rather cannot be changed for the good?

hubparticipant, 2021-02-02:
This is a big problem. Is this being handled in the latest beta? 15-20ms is kind of a big deal

hubparticipant, 2021-02-02:
> Ps: if it's the audio buffer I suppose this rather cannot be changed for the good?

If that is the case, a per track delay compensation is the only way to somewhat remedy this. 
That being said, I think a fairly limited hardware sequencer as the Polyend Tracker should be nice and tight

bnckd, 2021-02-02:
already files this here over a month ago ;) 
https://github.com/polyend/TrackerIssues/issues/573

NoraCodes, 2021-02-02:
> This is a big problem. Is this being handled in the latest beta? 15-20ms is kind of a big deal

> > Ps: if it's the audio buffer I suppose this rather cannot be changed for the good?
> 
> If that is the case, a per track delay compensation is the only way to somewhat remedy this.
> That being said, I think a fairly limited hardware sequencer as the Polyend Tracker should be nice and tight

I mean, the Tracker also does sampling, synthesis, effects... it's certainly not a dedicated MIDI sequencer. Also, while it may be very significant for you, it actually doesn't affect my use case at all - I think we should take such issues as something that is being worked on, and only buy the product if and when it fits your use case.

bnckd, 2021-02-02:
well, the use would be that of a drummachine / sampler and besides being a nice scratchpad it is a superconvenient solution for live use, but live you'd probably also have a dedicated synth or another drummachine or whatever needs sync. and if the audio is always late, it definitely is a very significant issue, and to be honest when i buy a product that belongs to the electronic side of the gear universe where sync / timing / quantization is one of the most important things i definitely expect it to work.

bnckd, 2021-02-02:
> > Ps: if it's the audio buffer I suppose this rather cannot be changed for the good?
> 
> If that is the case, a per track delay compensation is the only way to somewhat remedy this.
> That being said, I think a fairly limited hardware sequencer as the Polyend Tracker should be nice and tight

already filed this here over a month ago ;) https://github.com/polyend/TrackerIssues/issues/573

hubparticipant, 2021-02-02:
The product is advertised as a an old school tracker. These are tight and accurate. That’s kind of the point of a tracker. Therefor sloppy timing is the last thing you’d expect. 
It took me two days to find this information. 
20 milliseconds is a big deal

bnckd, 2021-02-02:
> I mean, the Tracker also does sampling, synthesis, effects... it's certainly not a dedicated MIDI sequencer. Also, while it may be very significant for you, it actually doesn't affect my use case at [...]

well, the use for most is probably that of a drummachine / sampler and besides being a nice scratchpad it is a superconvenient solution for live use, but: live you'd probably also have a dedicated synth or another drummachine or whatever else needs sync. and if the audio is always late, it definitely is a very significant issue, and to be honest: when i buy a product that belongs to the electronic side of the gear universe where sync / timing / quantization is one of the most important things (and imo not just something that can be worked on along the way...) i definitely expect it to work.

gurkenkimchi, 2021-02-02:
> The product is advertised as a an old school tracker. These are tight and accurate. That’s kind of the point of a tracker. Therefor sloppy timing is the last thing you’d expect.
> It took me two days to find this information.
> 20 milliseconds is a big deal

welp. even my old .mod/.it won't load properly. definitely not as described on the site.

hubparticipant, 2021-02-02:
> welp. even my old .mod/.it won't load properly. definitely not as described on the site.

I don’t know what welp means. 

Backwards compatibility can be a tricky thing. 

The fact of the matter is that 20 milliseconds of latency is huge. And in 20 years I’ve never seen this with hardware gear. Ever. 
MIDI drift, notes shifting around etc: yes. But 20ms of latency: never.

hubparticipant, 2021-02-02:
> I don’t know what welp means.
>
> Backwards compatibility can be a tricky thing.
>
> The fact of the matter is that 20 milliseconds of latency is huge. And in 20 years I’ve never seen this with hardware [...]

It is one of the reasons people use hardware. To get no latency. 
Squarepusher is probably the example as he refuses to use software because of latency issues

bnckd, 2021-02-02:
> welp. even my old .mod/.it won't load properly. definitely not as described on the site.

off topic, but i just had a jar of "oi kimchi" for breakfastlunch :D

ambv, 2021-02-02:
We have this problem on our radar. As I described elsewhere this is a complex issue that DAWs like Renoise and Ableton Live deal with in multiple ways.

If you're using Tracker as the master clock, essentially the question we're after is:
> If you have "C-4 M01" on one track and "C-4 01" on another track during the same step, do they sound at the same time?

The answer here depends on many factors which is why we don't have this yet in Polyend Tracker. At this point you need to compensate latency on the MIDI receiver end. On that same picture above you see that the "MIDI Clock Master" section in Renoise also has an Offset slider. That slider isn't all that you need to consider.

* If the MIDI out that you're sending to is a synth with zero latency, the "C-4 M01" note will sound sooner than "C-4 01" because Tracker's audio engine has some internal latency due to its audio buffer it needs for its synthesis engines, reverb, master limiter, and our hardware codec's "bass boost" and "space enhancer".
* However, that synth might either have latency of its own, or it can be chained to a pedal or other processor that adds some latency of its own. So calculating this latency automatically isn't possible to figure out.
* Better yet, the question arises: how are you listening to your external synth sequenced by Tracker? Are both Tracker and your external synth mixed by some other mixer or DAW? Or is the external synth's sound engine routed back to Tracker via its "Line In" audio input? In the latter case, the sound from the synth is additionally shifted by Tracker's audio buffer.
* Finally, the kicker: what if you have more than one device connected to MIDI? "C-4 M01" (MIDI channel 1) might have different latency from "C-4 M02" (MIDI channel 2).

Again, is this a Polyend Tracker-specific problem? Nope. Let's see how Renoise solves this. Apart from the "Offset" slider we've seen already, it's got "automatic plugin delay compensation". It can do this because VST plugins report their latency to the VST host. Tracker works with hardware and this kind of reporting is impossible.
[image]

On top of this, Renoise has an instrument-specific MIDI OUT latency setting:
[image]

That's not all. To deal with the situation of audio from the external MIDI synth being routed back to Renoise, it's got a special mode in its #Line Input device:
[image]

---

We are working on implementing the equivalent compensation on Tracker for the 1.4 firmware series. Our internal audio buffer is 128 samples long which at 44,100 Hz is 3ms. Sometimes we need additional buffer space but it should be nowhere near 20ms. We're still investigating this side of the story.
 
Once we're done implementing this, we'll make a video demonstrating what the measuring process is for us and how you can replicate this yourself.

hubparticipant, 2021-02-02:
> We have this problem on our radar. As I described elsewhere this is a complex issue that DAWs like Renoise and Ableton Live deal with in multiple ways.
> 
> If you're using Tracker as the master clock, essentially the question we're after is:
> 
> > If you have "C-4 M01" on one track and "C-4 01" on another track during the same step, do they sound at the same time?
> 
> The answer here depends on many factors which is why we don't have this yet in Polyend Tracker. At this point you need to compensate latency on the MIDI receiver end. On that same picture above you see that the "MIDI Clock Master" section in Renoise also has an Offset slider. That slider isn't all that you need to consider.
> 
> * If the MIDI out that you're sending to is a synth with zero latency, the "C-4 M01" note will sound sooner than "C-4 01" because Tracker's audio engine has some internal latency due to its audio buffer it needs for its synthesis engines, reverb, master limiter, and our hardware codec's "bass boost" and "space enhancer".
> * However, that synth might either have latency of its own, or it can be chained to a pedal or other processor that adds some latency of its own. So calculating this latency automatically isn't possible to figure out.
> * Better yet, the question arises: how are you listening to your external synth sequenced by Tracker? Are both Tracker and your external synth mixed by some other mixer or DAW? Or is the external synth's sound engine routed back to Tracker via its "Line In" audio input? In the latter case, the sound from the synth is additionally shifted by Tracker's audio buffer.
> * Finally, the kicker: what if you have more than one device connected to MIDI? "C-4 M01" (MIDI channel 1) might have different latency from "C-4 M02" (MIDI channel 2).
> 
> Again, is this a Polyend Tracker-specific problem? Nope. Let's see how Renoise solves this. First off, there's an "Offset" slider
> for clock information, both master and slave:
> [image]
> 
> It's also got "automatic plugin delay compensation". It can do this because VST plugins report their latency to the VST host. Tracker works with hardware and this kind of reporting is impossible.
> [image]
> 
> On top of this, Renoise has an instrument-specific MIDI OUT latency setting:
> [image]
> 
> That's not all. To deal with the situation of audio from the external MIDI synth being routed back to Renoise, it's got a special mode in its #Line Input device:
> [image]
> 
> We are working on implementing the equivalent compensation on Tracker for the 1.4 firmware series. Our internal audio buffer is 128 samples long which at 44,100 Hz is 3ms. Sometimes we need additional buffer space but it should be nowhere near 20ms. We're still investigating this side of the story.
> 
> Once we're done implementing this, we'll make a video demonstrating what the measuring process is for us and how you can replicate this yourself.

Ableton Live is probably the worst to use as an example/standard as it has the worst MIDI timing and has many clock sync issues. 
Renoise and Logic are solid. 

Anyway, it is nice to read this is at least on your guys radar.

NoraCodes, 2021-02-02:
> If the MIDI out that you're sending to is a synth with zero latency, the "C-4 M01" note will sound sooner than "C-4 01" because Tracker's audio engine has some internal latency due to its audio [...]

I want to say, also, that this issue is largely mitigated by routing audio through the Tracker, which also allows that audio to be resampled easily and to apply Tracker's EQ to that audio, which is extremely useful. So while this is definitely an issue, there is somewhat of a workaround available.

danapeks, 2021-02-05:
Hi great to hear this is on your radar for v1.4! Just hooked up to an external drum machine and using the Tracker as master transport and clock am also getting around 20ms delay which makes aligning drums on the tracker and the Roland just slightly out. Thanks!

bnckd, 2021-02-06:
> I want to say, also, that this issue is largely mitigated by routing audio through the Tracker, which also allows that audio to be resampled easily and to apply Tracker's EQ to that audio, which is [...]

if you mean workaround as in fixing it, then yes - if you mean workaround for now, then definitely no. there is no way to perfom in conjunction with other machines except you add a 500 euro erm multiclock.

gminorcoles, 2021-02-11:
I am researching known issues related to midi timing prior to purchasing the tracker because I want hardware sequencers to be tighter and more stable than what is achievable on the PC. I don’t think Renoise is a fair benchmark for latency and timing issues. An MPC 4000, Mc 909, Nerdseq, or other sequencer that has audio and midi processing, is fully featured, is a good benchmark. 

as for raw audio buffer latency triggering samples, the Squarp rample, the rosdum assimik8or are two sample players I can think of off the top of my head with 1ms audio playback latency or better. My PC gets latency of about 1 or 2 ms or even lower using the expert sleepers tool.

danapeks, 2021-02-11:
> I am researching known issues related to midi timing prior to purchasing the tracker because I want hardware sequencers to be tighter and more stable than what is achievable on the PC. I don’t think Renoise is a fair benchmark for latency and timing issues. An MPC 4000, Mc 909, Nerdseq, or other sequencer that has audio and midi processing, is fully featured, is a good benchmark.
> 
> as for raw audio buffer latency triggering samples, the Squarp rample, the rossum assimik8or are two sample players I can think of off the top of my head with 1ms audio playback latency or better. My PC gets latency of about 1 or 2 ms or even lower using the expert sleepers tool.

Good point. Be interesting to add the new Novation circuits or the Korg electribe 2 sampler into that mix if anyone has either of those?

joe-chip-cyber, 2021-02-19:
For those following along, I mentioned above that audio tracks in Logic Pro X do not have an audio delay setting.

This is true, but I had forgotten the built-in Sample Delay audio plugin has a ms delay setting. This seems to be a fairly light operation in CPU terms, and adding Sample Delays to incoming audio tracks (from Midi sound sources) is a fine work-around for the time being.

hubparticipant, 2021-02-21:
Yep sure 👍🏼 That is a possibility. You can always delay sound with trickery, but honestly the Polyend Tracker needs better latency. Period. It is just unaccaptable a hardware sequencer delays all my other sequencers-samplers-synths by almost 20 milliseconds. There is a reason I am still using stuff like the Akai S3000 and hardware synths. I do not accept latency on hardware, sorry.
This needs to be fixed ASAP.

NoraCodes, 2021-02-21:
I'm not sure exactly what you're suggesting should be implemented, @hubparticipant. A certain amount of audio buffering in required for the effects offered by the tracker. As a sequencer, it has no latency; by adjusting delay compensation where appropriate, it's possible to work around it. Even digital reverb pedals have latency.

bnckd, 2021-02-21:
and i'm not entirely sure why you defend this obvious flaw ;) 

you can use it as a midi seq, yes. that's what i ended up doing but you can't use it live in hybrid (without a midi delay you'll never be able to use it that way) and that's why i selling mine for now. also ableton, my sl3 and a cirklon are way better midi sequencers. 

another way to use the PT (for me and probably a lot of others as well) would be using it as a scratchpad and when i get back to the studio i would want to swap and work on certain tracks in realtime, like using a modular bassdrum instead of a sample and using a proper synth instead of the wavetables...i simply can't do that because there's the damn latency. even without any processing in the tracker there's 20 to 25ms. 

i have been working with (a lot of) outboard gear like this - be it new, old, vintage, shitty or digital or itb - for more than 20 years now. shitty midi sync can be corrected with audio clocking but usually you'd deal with a few ms here and to be fair i hardly had to use the erm and sold it at some point. old sloppy boxes usually work after a few restarts and you can definitely work whit a latency of a few ms and afterwards correct it in ableton. i mean i get a max of 2 to 3 ms latency with a 32 channel desk resp. 32 ins and 16 outs going through mixers, effects, patchbays and whatnot - i can listen to and process (almost) everything in realtime without any latency it works perfectly. the tracker was the first device in a long time i could not use in this setup at all and that's pretty sad tbh.

hubparticipant, 2021-02-21:
Stop defending this Nora. Please. You have no idea what you are talking about. This latency is simply not acceptable. 
And as the poster before me, I am too considering returning it or selling it. 
I bought this as soon as it had the ability to program MIDI chords. To use as a master sequencer. Which is just not possible unless you sacrifice 20ms of latency. Which in my experience I have never seen on any sequencer. Even the old cheap soubdcards had latency. 
This actually feels like working with a 1999 Soundblastee on Windows ‘98. 
The latency is THAT bad. 
Again, this needs fixing ASAP

NoraCodes, 2021-02-21:
Can I ask if you have any examples of a device which does sampling, reverb, delay, and master compression with zero latency? I agree that they should permit turning off the master fx to remove the delay, but I'm not entirely sure how you intend for the unit to do compression without latency.

hubparticipant, 2021-02-21:
Pretty much all Akai samplers up until 2010 (never used the new ones).
 Elektron’s. Yamaha’s, Roland,...
All of them. C’mon
This question of yours shows your ignorance. Exactly what I said earlier.

bnckd, 2021-02-21:
ähm...digitakt? octatrack? mpc (and mpc is know for midi issues...)? i have numerous modular devices with compressors and there is no delay. if due to the architecture this is or would or was never possible, than this is something they should write next to the specs, as this is probably one of the most important things to a lot of people. i do have the latency  all the time, with no master fx. even a digitakt send to the daw and back via ext ous and back in has almost no latency.

NoraCodes, 2021-02-21:
This is interesting to me as one of the biggest complaints I've seen online about the DT is regarding MIDI latency. But maybe everyone talking about that just doesn't know what they're talking about? 🙄

hubparticipant, 2021-02-21:
User error. They don’t have latency. Overbridge has. 
This, however, is about the Tracker. Not the Digitakt. 
And the Tracker needs a fix.

NoraCodes, 2021-02-21:
Yeah I mean I don't disagree with you that the latency should be reduced, but holy shit, get off your high horse. Issues you have are fundamental problems, but issues others have are user error? You're a parody of yourself.

hubparticipant, 2021-02-21:
Yes of course I sound like a king yelling on a horse. 
Because this is important and needs fixing. 

Look:
Polyend makes sequencers. It is what they do. 
Sequencers are supposed to be tight. It is a fundamental part of a seq. And thus a fundamental part of Polyend.
therefore this issue is priority for a company who builds sequencers for a living.

NoraCodes, 2021-02-21:
Oh wow, you're a king now? King of what, assholistan?

Again: this technical issue should be solved, but your ego is a much more pressing problem.

bnckd, 2021-02-21:
he said "yes he sounds like a yelling king" just like you said and no different from it..so no need to get passive agressive here. speaking of pressing problems: you're constantly defending a serious flaw as ... come on it's not that fundamental, where-as in fact, it definitely is most likely so, period. we could have stopped aged ago at exactly that point. just like sequencers or rhythmmachines, grooveboxes, etc are THE fundamental backbone in electronic music, period. over and out :)

NoraCodes, 2021-02-21:
> "yes he sounds like a yelling king", just like you said and no different 

I literally never said that. Y'all really need to work on your reading comprehension skills.

hubparticipant, 2021-02-21:
Etimology. Look it up. Pls.

MetabaronCZ, 2021-02-21:
It is about a month without bugfix release. I hope they are doing a big rewrite :).

bnckd, 2021-02-21:
literally no, you didn't say king. but you said "get off your high horse" as in kings on high horses. really, splitting hairs like that? and then going straight to judging other people and assuming the ain't got reading comprehension skills...oh my. 
been there done that, and i can assume too: this is your daily, typical and highly unnecessary forum shit.

hubparticipant, 2021-02-21:
Bottom line is that TO ME the Tracker is a dream of a sequencer. Intuitive, fast, simple. To the point. For both rhythmic content, melodic and harmonies. It is just brilliant. So I prefer to keep using it as there is no replacement afaik. 
The fact that the internal audio is 20ms late just ruins all of this functionality. 
If this can’t be fixed, it is not a bug but instead it is a feature and should be written in the manual (which I read twice before purchase): “internal audio is 20ms late”; it should read.

hubparticipant, 2021-02-21:
And this topic is about “internal audio latency”. 
Nora, if this isnt a problem to you, then why are you here?

PITmast, 2021-05-07:
Hello Everyone

Well, it's hot topic but I believe the quite simple solution is possible.

I figured out it's possible to get quite good results changing song tempo and micro-move on midi triggers.
For example try with tempo 68 and 'm  8' on every midi note.
It's not ideal. The tempo must be changed because m-m has too low resolution.

I found Clock sync delay in v1.4.0b1. My first thought was it is exactly what I did with micro moves but Clock sync delay in config doesn't work for me. I tried various values with various tempos and no difference at all. I hope I'm not using it correct.

Regards

theorize999, 2021-05-08:
I agree, there is a way to stress the importance of something without being an asshole. I've kind of pestered about a couple things a little bit but I don't ever remember turning into a dick about things. We've all got a priorities. I remember saying that tracking functions should take precedence over MIDI since the name of it is Tracker and all. And I use it with midi quite a bit so having tight timing IS indeed important to me. I sure as hell didn't throw a tantrum and start demanding they put in my tracking functions right now. Let's all just try to stay calm I think they are trying really hard to please everyone and I also think they are doing a pretty remarkable job at it. It's not like we didn't watch it and buy when we thought it had the features we wanted. Sorry if someone was wrong about that when they bought it but Polyend has steady improved this thing and at a pretty good rate, development has kept a good pace IMO.

theorize999, 2021-05-08:
@PITmast I can't promise it'll be today but as soon as I can I'll set up my drum machine and try it, I would like to get it integrated so I can have a few more channels for tracker to work with. If I get it to work I'll share my settings.

breskar, 2021-05-09:
I confirm having the same experience as well 1.4.0b1. Thanks for the micro move workaround PITmaster. I am using m14 on 130 bpm with not perfect but workable results.
Anyway I think that the master processing fx is playing a role here as I have way less delay when monitoring the signal with the line in.

PITmast, 2021-05-09:
Yeah, these micro delays on midi for compensation should be dynamic, of course, and be the function of tempo and audio latency at least. So the challenge is to count transient latency amounts for this function

Regards

theorize999, 2021-05-13:
Has nobody tried the "offset" parameter in the instrument settings for a MIDI instument?!?!?! 
Works pretty well for me. 
I mean sure it's not plank scale resolution but hey I think it's definitely useful, and really the best solution if you're running more than one MIDI unit from a pattern, not everything has the same latency so just line it all up to the slowest in the chain and you should be good. 
You guys owe me the price of a mixer.

breskar, 2021-05-13:
A simple solution and indeed should work for multiple midi devices connected.  I will try later for sure thanks

doelman, 2021-05-13:
> Has nobody tried the "offset" parameter in the instrument settings for a MIDI instument?!?!?!
> Works pretty well for me.
> I mean sure it's not plank scale resolution but hey I think it's definitely [...]

Will definitely try, thanks! Is that tempo dependent or can you just leave it as soon as you've found the best setting?

breskar, 2021-05-13:
It should be timing dependent as in the manual there's a % indicated. 

 "Micro-move, slightly moves (nudges) forward the step position. from 0% to 100%"

So 0% is the original position and 100% the next step ?

Implementing a kind of fixed Micro-Move for each midi channel could be a solution ? Not BPM dependent and saved with the project / general settings ?

[6 more replies]

closed 2023-04-21 by MiroPoly