polyend/TrackerBetaTesting #571: Suggestion: Song mode -> OFF pattern enhancement

anemlife, 2020-12-24:
When doing song edits in song mode, I copy/remove a lot of patterns and when I leave a blank I have to go back in pattern mode to add an OFF note at the beginning of the blank so the previous samples stop there.

Two problems:

- First it takes a lot of time to do when you have to check for multiple empty patterns / tracks!

- Secondly, there's no info in song mode if there's an OFF command at the beginning of an empty pattern. 

What about a line or a cross as an indicator that a blank pattern contains an OFF? That would definitely make the workflow better for edits! Also what about a shortcut to add an OFF in a blank pattern in song mode?

Thanks

ketumatek, 2020-12-24:
Hi really nice idea...in renoise when a slot (a pattern) is in off there is a X on the pattern which is mute. 
As this you can easily know witch track is mute or not.
May be they can do the same in song mode...
they can put colors that will be nice too.
- 8 colors for 8 tracks 
A 9 color for a mute track
A 10 colors for empty track

anemlife, 2020-12-24:
Yeah definitely, a better design in general for the song mode would be great! You can get lost really fast with long structures at the moment

A color code would definitely be welcome :)

tilllt, 2020-12-24:
And cloning a pattern to a new slot would be essential

anemlife, 2020-12-25:
> And cloning a pattern to a new slot would be essential

You already can using the same shortcuts than in pattern mode! (copy/paste button etc...) And if you hold shift you can select multiple patterns at once.

But take care it's sometimes buggy when you try to copy to much patterns at once!

tilllt, 2020-12-25:
I know, but that's not what I mean. I mean "copy pattern and paste it to an empty slot", just in one click. Now, it's copy, select empty slot, paste. Three clicks.

anemlife, 2020-12-26:
Oh I get it now! Yeah this option would really help