polyend/TrackerBetaTesting #572: Tracker freezes when trying to load a game [1.3]

bnckd, 2020-12-24:
tried this 3 times, freezes straight away. 
the rest seems to work just like in the beta!

bambrose, 2020-12-24:
you're pressing the black buttons (the ones under the screen without legends) to start the game right ?

bnckd, 2020-12-24:
Of course, there's only one option, can't remember if it said load or select, I think the most right one underneath the screen.

bambrose, 2020-12-24:
There's no legend... I just clicked them until it worked.
the second one in from the right starts or pauses the game (alternates between these two states)

piotr3500, 2020-12-24:
There is a legend in the Games menu.

bnckd, 2020-12-24:
I select a game with the cursor up down and press load or whatever it says
on the right, freezes.

bnckd, 2020-12-24:
done, works, my bad. and how do you stop the game? ;)

closed 2020-12-24 by bnckd

bambrose, 2020-12-24:
no worries :)
soft reset to restart as far as I can establish (i.e. hit the power button)