polyend/TrackerBetaTesting #573: midi clock delay for individual channels / midi clock & note per individual channel

bnckd, 2020-12-24:
1) would be neat to individually choose which midi channels get note on off and / or clock only. 

2) if 1) works out - a delay ( + - ) just like the sync delay for the individual midi channels would be superb, so if a machine is just getting clocked by tracker (no matter which sync mode it is in) and not in timing, you can adjust the timing yourself - this would make the erm multiclock obsolete best case scenario.

phaelam, 2020-12-28:
perhaps the default should not have a sync delay. mine was sounding like shit...then I checked sync delay. It was set by default to -49ms. mine sounds perfect at 0 delay.

I think the default would be best suited at 0.

bnckd, 2020-12-28:
Default at 0 of course, mine was like that (and sounded shitty until the recent update) though, strange that yours was - 49?! The indivudal delay per channel / clock would be a great addition nevertheless, more flexibility needed here.

inigokennedy, 2021-09-26:
a big up vote for adjustable output delay for midi clock / transport. we need to be able to bring external kit into sync with tracker audio when using tracker midi clock out. @Jacknametrouble *essential* when clocking external drum machines from the tracker for example. @bnckd midi clock / transport are system real-time messages and don't have a channel assignment so it could only be a global offset per midi port not per channel.