polyend/TrackerBetaTesting #574: [1.3.0] SD Card (Tracker) no longer mounts via android file transfer

bambrose, 2020-12-24:
On all previous versions I could use Android File Transfer to access contents of SD card via the 'file transfer' item in the config menu.
This has stopped working on most recent beta (i.e. I am having to remove SD card and use USB reader provided)

Just-Peter, 2020-12-24:
I can confirm that Android File Transfer to access contents of SD card via the 'file transfer' is not working anymore. In 1.3.0b5 it was still working. I'm using a Mac Mini M1 with Big Sur 1.11

jaap3, 2020-12-24:
I've was able to get Android File Transfer to access the Tracker on MacOS 10.14.6, but transferring a (tiny) file doesn't work. Afterwards AFT stops seeing the Tracker and closes itself. Restarting File Transfer Mode a couple of time seems to make the Tracker available, but copying a file still doesn't work.

piotr3500, 2020-12-24:
I'm testing on osx via Command One program and it works fine.

Just-Peter, 2020-12-24:
That’s good to hear! But why doesn’t work Android File Transfer on the Mac anymore? It was working on 1.3.0b5, but not on 1.3.0. final. From: piotr3500Sent: Thursday, 24 December 2020 15:57To: polyend/TrackerIssuesCc: Just-Peter; CommentSubject: Re: [polyend/TrackerIssues] [1.3.0] SD Card (Tracker) no longer mounts via android file transfer (#574) I'm testing on osx via Command One program and it works fine.

Just-Peter, 2020-12-24:
@piotr3500 I don't see any MTP devices in Commander One when I choose 'start file transfer' on my Tracker. How did you do this?

ndguardian, 2022-01-18:
Just a heads up, looks like the feature was removed. Presumably there are plans to reimplement, but not certain. Just stumbled onto this support article about it. 

https://help.polyend.com/330743-Can-I-use-the-Media-Transfer-Protocol-MTP-to-copy-files-on-the-SD-card-using-direct-connection-between-the-Tracker-and-my-computer

closed 2023-04-26 by MiroPoly