polyend/TrackerBetaTesting #576: [Firmware version] Saving Project freeze , have to unplug and restart 1.30

billyboy1967, 2020-12-24:
## Hi, sorry you had trouble!

### Steps to reproduce

Please describe the steps to reproduce the bug. Tell us what you noticed
happening and what you expected to happen instead.

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once
Every time I save the project
* Does this bug happen in previous firmware versions?
Yes
* Does this bug happen in every project?
Yes

shuler323, 2021-01-04:
Please provide more info for us (project, videos etc) to take care of this issue. For now it is nearly impossible to reproduce.
Remember to note exact firmware version in title.

MortalJS, 2022-01-12:
Close for no response received.