polyend/TrackerBetaTesting #577: Single Instrument or/and steps as sidechain source (not entire tracks)

LetsSeeIfWeSolveThis, 2020-12-24:
Could bring a way **more surgical sidechain feature** with the same limiter.

**In beat slice mode, the step sidechain source** would be a great feature, since I usually use the different slices as different instruments.

tilllt, 2020-12-26:
How is side chain compression available "per track" now? I though it's just on the line-in?

math0ne, 2021-01-04:
> How is side chain compression available "per track" now? I though it's just on the line-in?

It's assignable in the limiter section of the master effects.

sunegg1024, 2021-01-15:
Was going to add this as a separate feature but figured it was better here. A cleaner solve for this, in my opinion, might be to have a Sidechain Source FX and be able to send whatever step to the limiter you would like. That would free you from losing a whole channel to sidechain and allow you to be more precise in what gets sent. Maybe the limiter level then can be applied as well with hex i.e. Placing an sc65 on step 1 of track 1 triggers the limiter for only that step and sets limiter to 65, ducking all other tracks. Just wish we could select which tracks are sidechained and which are not..